supermerill/SuperSlicer #975: Built in calibration generators produce errors about the placement of the object

brianredbeard, 2021-02-23:
### Version
Version 2.3.55 (Currently pre-release build `201c612` "Super Brim Ears", filename `SuperSlicer-ubuntu_18.04-2.2.53.5.AppImage` - see #973).

### Operating system type + version
Linux (Fedora 33, Kernel 5.10.13-200.fc33.x86_64, GNOME 3.38)

### 3D printer brand / version + firmware version (if known)
A wildly hacked Ender 3

### Behavior
The built in calibration generators currently produce errors about the placement of the generated object when run. This was discovered when attempting to generate a calibration cube and seems to occur with all combinations of settings.  To rule out local configuration issues, SuperSlicer was restarted with an empty profile the following presets were used to duplicate the issue:

  - Original Prusa i3 MK23S
  - Prusament PLA
  - 0.15mm OPTIMAL nozzle

After loading these settings, selecting `Calibration -> Calibration Cube -> Standard Cube`  produces an error in the lower right portion of the plate view.  Pressing the "standard cube" or "voron cube" button multiple times causes additional pop-up errors to occur.

[image]

[image]

[image]

These errors all disappear after the calibration cube dialog is closed.

supermerill, 2021-02-24:
seems like a gui refresh is missing, as rotation the view also remove it.

brianredbeard, 2021-02-24:
FYI, found some other "interesting" behavior.  This would make me think that there is something beyond a simple GUI refresh.

When taking the calibration cube and using the built in "cut" tool, the resulting artifacts are placed at the world coordinates `(0,0,5)` (with the 5mm "Z" value being half the size of overall height. I also confirmed that if the cut is changed to 16mm instead of 10mm that the resulting artifact is placed at `(0,0,8)`)

Before cut execution:

[image]

After cut execution (meaning, the only user operation which occurred was clicking the "perform cut" button):

[image]

brianredbeard, 2021-02-24:
Interestingly, performing a _second_ cut on the object works as expected.

qwewer0, 2021-09-12:
> When taking the calibration cube and using the built in "cut" tool, the resulting artifacts are placed at the world coordinates `(0,0,5)` (with the 5mm "Z" value being half the size of overall [...]

Still a problem on 2.3.57.0 (#938 too)

supermerill, 2021-09-27:
that's because the cube origin is at its center.
As nothing can go under the bed, it's raised just enough to touch the bed.

brianredbeard, 2021-11-15:
Closing this issue because the original concern ("Built in calibration generators produce errors about the placement of the object") have been resolved.

Thanks to the explanation by @supermerill my note about the location handling with the cut tool.  While it's still not what I would have expected it's unrelated to the initial parsing of calibration objects.

closed 2021-11-15 by brianredbeard