supermerill/SuperSlicer #976: How to influence curve segment length?

CCS86, 2021-02-24:
How can I change the segment length / max deviation /angular split for curves? 

[image]

[image]

supermerill, 2021-02-24:
If it's less defined than the stl, check:
  * printer->general->advanced -> min extrusion length
  * print->slicing->resolution (should be <= than 'min extrusion length')
If you want something more defined than the stl: print->slicing->curve smoothing, but it's a bit difficult(buggy?) to make it works right now.

CCS86, 2021-02-24:
Thank you! It was in fact min extrusion length forcing that facet size.

CCS86, 2021-02-25:
I did some testing and found that the default value for min. extrusion length of 0.035mm was causing significant surface finish issues for me. No combination of jerk and accel could clear the junction artifacts at 60mm/s. This is running RRF3.2 on a Duet Maestro, printing an oval vase (rescaled cylinder inserted in SS) 20mm x 15mm.

I tested at min. extrusion length = 0.01mm and found a big improvement. But needing segment lengths of only 0.0004" seemed odd to me. I looked at the gcode and found the average segment length was 0.62mm. 

Is this setting actually setting minimum segment length directly, or is it a maximum deviation value?

CCS86, 2021-03-03:
@supermerill 

I am still trying to understand how to control curve segmentation. After doing a bunch more testing, the "min extrusion length" really does seem to act more like an allowable deviation.

I am using added cylinders in SS to test, avoiding STL facet influence.

With a 0.8mm diameter cylinder (the smallest you can theoretically print w/ a 0.4mm nozzle), and "min extrusion length" = 0.008mm, I get gcode with average segment lengths of 0.075mm. Roughly 9.5x what that setting would suggest.

[0.8cylinder.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6076354/0.8cylinder.3mf.txt)

If I scale that cylinder up to 20mm diameter, and leave "min extrusion length" = 0.008, I get print segments that are 0.48mm long. For these tests I have "resolution" = 0 and "model rounding precision" = 0.00001mm. 

I'm curious if there is another setting governing this behavior. It seems like you would need at least two settings to have more complete control: one setting a max allowable deviation, and one setting a minimum segment length. 

Any insight would be appreciated!

supermerill, 2021-03-18:
I'll have to investigate.