supermerill/SuperSlicer #977: SuperSlicer_2.3.55.5_macos_20210223 does not start onmacOS Big Sur

AKlionRU, 2021-02-24:
SuperSlicer_2.3.55.5_macos_20210223 does not start onmacOS Big Sur

formadi, 2021-02-24:
Yeah me too. 
Look at the release notes it's commented that macOS 11 can't be run. 
I think I have to wait a bit.

AKlionRU, 2021-02-25:
It is not good!

nanoplane, 2021-02-25:
It's weird, I can build it on the mac and the runtime works fine..

alienboyxp, 2021-02-25:
> It's weird, I can build it on the mac (Big Sur 11.2) and the runtime works fine..

You can provide the dmg/app?

nanoplane, 2021-02-25:
> > It's weird, I can build it on the mac (Big Sur 11.2) and the runtime works fine..
> 
> You can provide the dmg/app?

No, I haven't figured out how to generate that yet..

nanoplane, 2021-02-25:
It seems that the test routines are failing to build

royal32, 2021-02-25:
> It seems that the test routines are failing to build

add the `-DSLIC3R_BUILD_TESTS=OFF` flag to your cmake command. @nanoplane 

I've compiled 2.3.55.5 for macOS and linked it below (too big to attach to a github issue) to use in the meantime until merill can get a working build released. (I hope thats OK merill, if not I'll delete it!)

[2.3.55.5 working macOS build](https://mega.nz/file/ZDpHnCrR#k0oCN5RW49RsQIS7_LHEBy7JHimjlo9NEJclegfg_18)

nanoplane, 2021-02-25:
So how are you getting it to build the mac app? (it's been about 20 years since I've done any real code development on the mac so... I may be a little rusty :-)

royal32, 2021-02-25:
I've described the problem here:

https://github.com/supermerill/SuperSlicer/issues/777#issuecomment-784623197

nanoplane, 2021-02-26:
not quite what I was after...  
I'me building in Xcode and trying to figure out what incarnations I need to do to get the .app package built and signed... is there a script there I'm missing or do I need to add a new target for the app and bring in the resources and binaries?  or is this a test? :-)

royal32, 2021-02-26:
Ah, misunderstood you. 
I used SuperSlicer's automated github action building scripts [here](https://github.com/supermerill/SuperSlicer/blob/master/.github/workflows/ccpp_mac.yml) to make the following script. 

```
#!/bin/sh

mkdir pack
mkdir pack/SuperSlicer
mkdir pack/SuperSlicer/SuperSlicer.app
mkdir pack/SuperSlicer/SuperSlicer.app/Contents
mkdir pack/SuperSlicer/SuperSlicer.app/Contents/_CodeSignature
mkdir pack/SuperSlicer/SuperSlicer.app/Contents/Frameworks
mkdir pack/SuperSlicer/SuperSlicer.app/Contents/MacOS

cp -Rf ../resources pack/SuperSlicer/SuperSlicer.app/Contents/Resources
cp pack/SuperSlicer/SuperSlicer.app/Contents/Resources/icons/slic3r.icns pack/SuperSlicer/SuperSlicer.app/Contents/resources/Slic3r.icns
cp src/Info.plist pack/SuperSlicer/SuperSlicer.app/Contents/Info.plist
echo -n -e 'APPL????\x0a' > PkgInfo
cp PkgInfo pack/SuperSlicer/SuperSlicer.app/Contents/PkgInfo

cp -f src/SuperSlicer pack/SuperSlicer/SuperSlicer.app/Contents/MacOS/SuperSlicer
chmod u+x pack/SuperSlicer/SuperSlicer.app/Contents/MacOS/SuperSlicer
tar -cvf SuperSlicer.tar pack/SuperSlicer

hdiutil create -ov -fs HFS+ -volname "SuperSlicer" -srcfolder "pack/SuperSlicer" temp.dmg
hdiutil convert temp.dmg -format UDZO -o SuperSlicer.dmg
rm temp.dmg
```

i put it in the build folder and call it pack.sh. that should take the compiled executable and pack it in an a .app and pack that in a .dmg

nanoplane, 2021-02-26:
Ah, it just builds it as an unsigned app, ok

ZhuJo, 2021-02-27:
Not working here too, Big sur 11.2

ZeekWorld, 2021-03-12:
Same, not working on Big Sur 11.3 Beta

brettvitaz, 2021-03-15:
FWIW, after the latest code push, I was able to build successfully and this problem is resolved for me. 

The prerelease build in Releases does not work for me.

gretel, 2021-03-17:
> FWIW, after the latest code push, I was able to build successfully and this problem is resolved for me.
> 
> The prerelease build in Releases does not work for me.

@brettvitaz can you share your build for testing please?

brettvitaz, 2021-03-18:
Here is my build: https://gofile.me/4tJmr/4MJF5TDs3

jremen, 2021-03-18:
Does it work on Catalina too or one has to download to separate versions? Because the Big Sur build I have from other user won't work under Catalina, only on Big Sur I use on my main computer.
I wasn't successful to build Superslicer at all, no problem with PrusaSlicer…

LazaroFilm, 2021-03-19:
> Here is my build: https://gofile.me/4tJmr/4MJF5TDs3

Why isn't this build in the official release?

brettvitaz, 2021-03-19:
@jremen I don't know if it works on Catalina. I don't have the version of MacOS on a machine readily available for testing.

supermerill, 2021-03-20:
@LazaroFilm 
> Why isn't this build in the official release?

I don't have a macos computer/vm to test.
here is the probably working [official nightly](https://github.com/supermerill/SuperSlicer/suites/2281540987/artifacts/47710605)

btw, the thread about this Aussie should be #777

jremen, 2021-03-20:
> @jremen I don't know if it works on Catalina. I don't have the version of MacOS on a machine readily available for testing.

So I can confirm brettvitaz build of SS 2.3.55.5 works on Catalina and Big Sur as well.

[image]

supermerill, 2021-03-20:
@brettvitaz what i have to do to create a good dmg?

nanoplane, 2021-03-20:
Build it for release

target=$pwd
cd deps
mkdir -p build
cd build
cmake
make

# copy the missing lib
cd destdir/usr/local/lib
cp libwxscintilla-3.1.a libwx_osx_cocoau_scintilla-3.1.a
# cd back to the base directory
cd $target
#make the build directory
mkdir -p build
cd build
#build it all
cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
make -j4

Then when done, execute the pack.txt script from the same build directory (magic happens)..
[pack.txt](https://github.com/supermerill/SuperSlicer/files/6176583/pack.txt)

supermerill, 2021-03-21:
I think it's pretty what i do to build already. What i mean, is what I have to modify in the codebase (commits) to make that to create a good compiled code.

sslupsky, 2021-03-27:
Something doesn't look right in the 2.3.55.5 screenshot?  The "paint on supports" icon appears to be missing?  Has 2.3.55.5 been merged with the release version of PS 2.3.0?

[image]

supermerill, 2021-03-27:
@sslupsky ??
[image]

sslupsky, 2021-03-29:
Yeah, that is my screenshot of PrusaSlicer 2.3.0.  The screenshot earlier in this thread does not have this icon?

commit ee4f2cf5 (2018-08-03): Try to fix #977
closed 2021-06-04 by supermerill