supermerill/SuperSlicer #978: Flow Calibration doesn't disable Ironing Post Process

BenWhite27, 2021-02-25:
### Version
2.3.55.5

### Operating system type + version
Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
Modified CR-10S running Klipper.

### Behavior
When performing a flow calibration the new Ironing Post Process isn't disabled in the printer profile.  This results in ironing the top layer twice, once with the Ironing infill option and another time with the post process.

[image]

As this calibration is intended to work with the original ironing option it might be best to alter the Print Settings for the print.

#### Project File (.3MF) where problem occurs
[filament_flow_test.zip](https://github.com/supermerill/SuperSlicer/files/6042317/filament_flow_test.zip)

ndaman, 2021-11-04:
I've been following SupeerSlicer's calibration process and I'm curious if this is causing an issue. The flow that looks the best on the top is quite low (~75%) and it leaves visible gaps in the first layer. Is there any way to manually fix this or should I use a different calibration method in the meantime?

supermerill, 2021-11-04:
It's an issue, as the ironing post-process will hide top defects used to evaluate the flow.

closed 2021-11-19 by supermerill

reinux, 2023-04-02:
I'm having this issue in 2.5.59. I'm not able to disable it manually either.