supermerill/SuperSlicer #979: Thumbnail sizes are not saved in the ini file

raidolo, 2021-02-25:
### Version
2.3.55.5

### Operating system type + version
Windows 10 1809

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
Thumbnails sizes are not saved in the profile.

In the .ini files I can see these entries:

```
thumbnails = 0x0,0x0
thumbnails_color = #018aff
thumbnails_custom_color = 0
thumbnails_with_bed = 1
```

While in the GUI I've saved 48x48 and 300x300 thumbnail sizes. Of course if you switch the profile, they switch back to 0, 0, cause there's nothing in the profile file.

I've also deleted all the physical printer profiles and started from scratch to avoid any incompatibility with previous settings, but I had no luck. 

#### Project File (.3MF) where problem occurs
Not relevant.

unkn0wndude, 2021-02-26:
I can confirm this. The workaround for now is to set them manually in the .ini file.

raidolo, 2021-02-26:
> I can confirm this. The workaround for now is to set them manually in the .ini file.

What's the syntax? 

thumbnails = 48x48,300x300 ?

supermerill, 2021-03-17:
as Prusa has implement their own version, but with a different gui method, there was a silent merge bug.

closed 2021-06-04 by supermerill