supermerill/SuperSlicer #982: Gap Fill not consistent

raidolo, 2021-02-26:
### Version
2.3.55.5

### Operating system type + version
Windows 10 1809

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
Gap Fill is not done if the Solid Infill Area Threshold is lower than the default value (70)  
If I set the Solid Infill Area Threshold to 1, the layer has the standard infill, but there's no gap fill where it "should" be. 
Instead SS tries to use the infill also in the gap, and depending on what infill is in use, the part looks more or less filled.

* _Expected Results_
[image]

* _Actual Results_

[image]

#### Project File (.3MF) where problem occurs
It's a benchy :) but still...here it is....

[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/6050743/3DBenchy.zip)

supermerill, 2021-03-18:
In the first pic, it's the gap fill from the solid infill.
If you only want 2 perimeters, no need to add more gapfill in-between because the expected wall width is already acheived.

If you don't want a solid layer (that's why you reduced the 'Solid Infill Area Threshold', right?), I don't see what is wrong on the second pic: it has 2 perimeters and a sparse infill

supermerill, 2021-03-19:
that's a new feature, with a new gap-fill threshold.