supermerill/SuperSlicer #984: One Perimeter on Top Surface Detection Failing

BlueDragonX, 2021-02-28:
### Version
2.3.55.5

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Voron v2.4

### Behavior
The "one perimeter on top surface" detection is broken for the [80t gear](https://github.com/VoronDesign/Mobius-Extruder/blob/master/STLs/%5Ba%5D_80t_gear.stl) on the Voron M4 extruder. Enabling this causes the gear teeth to only have a single perimeter. The rest of the walls have the appropriate number of perimeters. Disabling this feature fixes the problem. Pic:

[image]

#### Project File (.3MF) where problem occurs
[[a]_80t_gear.zip](https://github.com/supermerill/SuperSlicer/files/6057792/a._80t_gear.zip)

CCS86, 2021-02-28:
I notices this too, where that setting causes a single external perimeter and sparse infill, when it should be the normal amount of perimeters.

BlueDragonX, 2021-03-11:
I've been able to reproduce this on a number of other STL files. Any idea what might be causing this?

closed 2021-06-04 by supermerill