supermerill/SuperSlicer #985: [2.3.55.5] Gap filling doing some bizarre stuff

CCS86, 2021-02-28:
2.3.55.5 / Win10 x64

I'm not even sure how to describe this. Gap filling is doing some non-optimal work between walls and bottom solid infill:

[image]

[Head V3.5a.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6058402/Head.V3.5a.3mf.txt)

cypeks, 2021-03-17:
Me too:
2.3.55.5:
[image]

and this same profile 2.3.55.2:
[image]

cypeks, 2021-03-18:
Yes, disabling "Only one perimeter on Top surfaces" helped.