supermerill/SuperSlicer #987: Feature request: Flow exponent multiplier

burner-, 2021-03-01:
CNC kitchen made good video what shows how flow multiplier is not constant at all when we go with bigger flows
[image]
https://www.youtube.com/watch?v=xAiUZ7HDhdM&t=820s&ab_channel=CNCKitchen

Specially with petg I run lot of these issues that slow extrusions get over extruded and fast ones get under extruded even there is no skip at driver. I already use volcano nozzles to decrease this problem but I think this is something what needs own compensation parameter to slicer filament settings. Basically this is possible to solve by adding exponent parameter for flow multiplier.

pasiz, 2021-03-01:
If your print goes to overextrusion, you have something terribly wrong with your printer. You cannot extrude more than 100% of feeded filament. So extruder calibrated badly or filament is bigger in real world than calculations... I run mostly 100-200mm/s with klipper and don't see problem like this. Just fix your volumetric feedrate to amount your hotend is capable of or get better hotend with more heating area. There are also multi heat channel nozzle hotends on market, volcano is single channel...

burner-, 2021-03-02:
@pasiz 
You miss point. Thing is that if I calibrate it perfectly for slow print rates then it under extrude at fast rates. So without exponent calibration is always compromise. And that hotend cabability is not binary factor as we see from CNC citchen tests. It start slowly drop almos immediately and it is just matter of what is acceptable tolerance. As always with slicers there is lot of things what we can resolve by going slower. But if we can get about 2x more cabability from same hardware with one tuning parameter why to not do it? Currently I run about 15mm3/s with low quality prints but with this I can push about 20-30mm3/s.

pasiz, 2021-03-02:
If you push more filament than your nozzle is melting, you are generating pressure between extruder and nozzle, thus when slowing down you see massive overextrusion when this pressure purges. In my humble opinion that option would be the best on firmware side, based only on fact that slicer doesn't know exactly your acceleration / deceleration ramps nor exact speeds on every point in real world, especially when using firmwares like klipper. I understand your idea and also myself have struggled with that. I would like to see that kind of feature in klipper, maybe worth of coding.

burner-, 2021-03-02:
At least in marlin there is linear advance for compensating pressure build in. You maybe right that this might be more like firmware feature than slicer thing.

pasiz, 2021-03-11:
Thinked this few days before commenting...

As said, you cannot purge your pressure build up programmatically, if your extruder is not outputting what your extruder motor gives in. So it is completely bullshit that your extrusion ratio varies; plastic is placed on wrong place in reality. So every time you slow down, extra plastic is put (amount your extrusion performance lowers on move before that). So you need to give more pressure when going over your melt limit and purge pressure when you are going on sweet spot after that. So your pressure advance in reality should be nonlinear, thus dependable on extrude volume / s, logarithmic on time scale after acceleration. Not easy task to solve, easier and more efficient would be shopping better hot end...

supermerill, 2021-03-17:
I can do something about it... but it won't be good because the slicer isn't the one that dictates the flow rate but the firmware.
The slicer output the desired moves, as if there is an infinite acceleration.
I can't increase/decrease the flow rate because I can't know when and where the speed is, as I have no control on when it accelerates, and if it really uses the acceleration parameter I pass on.
It's a bit like if I implement a pressure advance in the slicer side, it will be sub-par.

pasiz, 2021-03-17:
https://github.com/KevinOConnor/klipper/blob/master/docs/Kinematics.md (last chapter) shows exactly, how it's implemented in firmware side of klipper. But as hookes and poisseuille would work when there is no change in viscosity caused by thermal conduct, this needs really complex tuning loop and still gives quite unpredictable results. In marlin / reprap style firmware this could some level be done in slicer, but in complex firmwares effect is quite nothing or even worse for physical outcome.

burner-, 2021-03-17:
Actually PA does not add extra flow at Klipper or Marlin. It just advance it. Not releated to this but it is interesting to see how different marlin and klipper implementation from that same thing is. At Klipper looks that printing speed control purely E speed so if printing speed is high enough E motor will skip. At Marlin E jerk will limit max PA speed and if printing speed request more PA than E jerk limit can allow it will limit printing speed. 
Anyway it looks that at both it will only modify time axis of extrusion instead of adding any factor by flow rate.

pasiz, 2021-03-18:
> Actually PA does not add extra flow at Klipper or Marlin. It just advance it. Not releated to this but it is interesting to see how different marlin and klipper implementation from that same thing [...]

I tried to explain that if you add extra plastic, it would just be on the wrong place, volume to extrude is same, fast or slow. So only time domain changes to feed pressure are possible, to the point where your extruder cannot push filament through. I print pla @ 160mm/s 0,4mm with klipper and have no skipping at extruder, but maybe you need something extraordinary. How fast you are needed to print and how big nozzle?

Sorry offtopic supermerill

burner-, 2021-03-27:
After calibrating PA(LA) I like to say that yes it is first level for everything. Anyhow it wont fully fix this with large volumes. With 30mm3/s everything is still ok. Probably it will be possible to squeeze even 40mm3/s with this feature but as you say probably better place for it is frimware. Lets put perspective. If you print 160mm/s 0.4 nozzle and 0.2 layer it is 12mm3/s so we speak quite different scale numbers here. I use 0.8mm nozzle. 0.4mm layer and with that problems start somewhere at 90-100mm/s. With smaller nozzle or thinner layer I naturally can go faster but it wont affect to total print time

pasiz, 2021-03-28:
As said, PA cannot cope with viscosity change in melting chamber, so that would need completely new mechanism that is conscious of melt rate, change of viscosity in volume / time domain / heat block temp, so the heat exchange between hotend / filament needs to be compensated and extrude volume should be calculated again. Completely different story and way beyond slicer capability. To problem easiest solve is to change extruder to multi chamber type, not e3d cheap type. 40mm2/s is nothing achievable with e3d (trust me, i have tried) if layer adhesion needs to be some reasonable level. You could try that theory by generating cylinder, put slicer to vase mode and let extrude large continuous arc with over 40mm2/s, you have to tune your temperature also much higher. In my point of view that is shoehorning expensive extruder capabilities to cheap one, with software. Laws of physics are still there and cannot be beaten easily. I print mostly abs and asa, and 40mm2/s is not achievable even close with those stuffs.