supermerill/SuperSlicer #988: Angel cost and Travel cost affect to rear seem position

FillPavS, 2021-03-01:
### Version
2.3.55.5

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
UlTi Steel / klipper

When choosing the position of the rear seam, the parameters Angel cost and Travel cost continue to affect, although they are inactive. If I leave there the default setting (100), the seam is not always going to be back, but if I switch to cost-Based and to set the values in these fields to 0, and then click rear seam position, he begins to work as it should.

https://www.youtube.com/watch?v=yG7l7GpFwaU

[image]
[image]
[image]
[image]

supermerill, 2021-03-17:
Can't reproduce, please give me your (zipped) project file.

FillPavS, 2021-03-18:
> Can't reproduce, please give me your (zipped) project file.

if still relevant
[seam_position_bug.zip](https://github.com/supermerill/SuperSlicer/files/6161356/seam_position_bug.zip)

in this version the problem is solved - https://github.com/supermerill/SuperSlicer/actions/runs/661956829

Thanks for your work

closed 2021-03-18 by FillPavS