supermerill/SuperSlicer #989: Start Gcode problem

misiek271, 2021-03-01:
### Version
2.3.55.5
### Operating system type + version
win 10 64Bit

### 3D printer brand / version + firmware version (if known)
_hevo

### Behavior
new version dont wait for heat up the bed end nozzle its start printing just after click print. previous version was ok it heat the nozzle and bed before make any print. i check the same settings in old and new version

nanoplane, 2021-03-02:
This line in code.cpp

https://github.com/supermerill/SuperSlicer/blob/201c6120b6c6122843ea04a388c3d93eb9cce1b7/src/libslic3r/GCode.cpp#L2166
 
needs to have the "wait" parameter to set_temperature changed to true.

anomalchik, 2021-03-04:
I have same issue. My printer have Marlin 2.0 and custom gcode runs before M109 and M190 command.

Pevious Release
`M140 S75; set bed temperature
M104 S232; set temperature
;TYPE:Custom
; custom gcode: start_gcode
G28 ; home all axes
G1 Z30.0 F9000 ;move the gentry up 30mm
; custom gcode end: start_gcode
M109 S232 ; set temperature and wait for it to be reached
M190 S75 ; set bed temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; custom gcode: start_filament_gcode
; Filament gcode
G1 Y50.0 F3600 ;Move bed to clip
G92 E0 ;zero the extruded length
G1 E13 F200 ;extrude 13mm of feed stock
G1 Z2.8 F9000 ;move the gentry down to clip
G1 X20 F3600 ;move x axis through clip
G92 E0 ;zero the extruded length again
G1 F9000
; custom gcode end: start_filament_gcode
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2`

2.3.55.5:
`M140 S75 ; set bed temperature
M104 S232 ; set temperature
;TYPE:Custom
G28 ; home all axes
G1 Z30.0 F9000 ;move the gentry up 30mm
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
G1 Y50.0 F3600 Move bed to clip;
G92 E0 ;zero the extruded length
G1 E13 F200 ;extrude 13mm of feed stock
G1 Z2.8 F9000 ;move the gentry down to clip
G1 X20 F3600 ;move x axis through clip
G92 E0 ;zero the extruded length again
G1 F9000
M109 S232 ; set temperature and wait for it to be reached
M190 S75 ; set bed temperature and wait for it to be reached
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2`

misiek271, 2021-03-09:
im ussing alredy compilated version. Can you change that to compiled version?

supermerill, 2021-03-17:
add the wanted behavior in the start gcode window

closed 2021-06-04 by supermerill