supermerill/SuperSlicer #991: [2.3.55.5] Number of perimeter is not respected

elpopo-eng, 2021-03-03:
### Version
2.3.55.5

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Number of perimeter is not respected in the 2.3.55.5 compared to 2.3.55.3

[image]

#### Project File (.3MF) where problem occurs
[Pb_Perimeter.zip](https://github.com/supermerill/SuperSlicer/files/6074509/Pb_Perimeter.zip)

Pavulon87, 2021-03-03:
Disable "Only one perimeter on Top surfaces" and try again.

elpopo-eng, 2021-03-03:
It works with one perimter on top surface disabled.

linux-paul, 2021-03-09:
Also in my case
[image]

closed 2021-06-04 by supermerill