supermerill/SuperSlicer #993: [2.3.55.5] Force Close on Mac

dwisetiyadi, 2021-03-04:
### Version
2.3.55.5

### Operating system type + version
Mac OS BigSur 11.2.1

### 3D printer brand / version + firmware version (if known)
DIY 3D Printer with Klipper

### Behavior
Always force close when i open superslicer. But looking good on previous version.

#### Project File (.3MF) where problem occurs
Can not save any file, because, it close when opened.

dianlight, 2021-03-06:
Same on my environment.

**Operating system type + version**
Mac OS BigSur 11.2.2 ( Apple M1 )

This is the Problem Report:

```
Process:               SuperSlicer [55873]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.3.55.5 (SuperSlicer_2.3.55.5)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [55873]
User ID:               501

Date/Time:             2021-03-06 10:36:30.747 +0100
OS Version:            macOS 11.2.2 (20D80)
Report Version:        12
Anonymous UUID:        F3A22DA5-EB7B-6F48-C732-8B9C0B22A131

Sleep/Wake UUID:       2A922592-A2A5-4766-B5AA-6BCE8E026226

Time Awake Since Boot: 590000 seconds
Time Since Wake:       41000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [55873]

VM Regions Near 0x8:
--> 
    __TEXT                      104dba000-106b26000    [ 29.4M] r-x/r-x SM=COW  /var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x000000010650b94b wxMacCoreGraphicsPenBrushDataBase::CalculateShadingValues(void*, double const*, double*) + 59
1   com.apple.CoreGraphics        	0x00007fff24eccaf0 CGFunctionEvaluate + 251
2   com.apple.CoreGraphics        	0x00007fff24ecc94a function_evaluate + 358
3   com.apple.CoreGraphics        	0x00007fff24ecc0d7 ripc_AcquireFunction + 974
4   com.apple.CoreGraphics        	0x00007fff24ecaffb ripc_DrawShading + 7518
5   com.apple.CoreGraphics        	0x00007fff2524c2b6 CG::DisplayListExecutor::drawShading(CG::DisplayListEntryShading const*) + 894
6   com.apple.CoreGraphics        	0x00007fff25305f62 CG::DisplayList::executeEntries(std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*, bool) + 228
7   com.apple.CoreGraphics        	0x00007fff24f5ead9 CG::DisplayList::execute(CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*) + 363
8   com.apple.CoreGraphics        	0x00007fff24f5e6d2 CGDisplayListDrawInContextDelegate + 701
9   com.apple.QuartzCore          	0x00007fff26e850e3 CABackingStoreUpdate_ + 544
10  com.apple.QuartzCore          	0x00007fff26eea141 invocation function for block in CA::Layer::display_() + 53
11  com.apple.QuartzCore          	0x00007fff26e844c4 -[CALayer _display] + 2183
12  com.apple.AppKit              	0x00007fff22d44bb1 -[_NSBackingLayer display] + 475
13  com.apple.AppKit              	0x00007fff22cb00f5 -[_NSViewBackingLayer display] + 555
14  com.apple.QuartzCore          	0x00007fff26e834c8 CA::Layer::display_if_needed(CA::Transaction*) + 874
15  com.apple.QuartzCore          	0x00007fff26fc35fb CA::Context::commit_transaction(CA::Transaction*, double, double*) + 517
16  com.apple.QuartzCore          	0x00007fff26e64e95 CA::Transaction::commit() + 783
17  com.apple.AppKit              	0x00007fff22d59e22 __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
18  com.apple.AppKit              	0x00007fff234a8ec2 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
19  com.apple.CoreFoundation      	0x00007fff204816d9 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
20  com.apple.CoreFoundation      	0x00007fff20481575 __CFRunLoopDoObservers + 549
21  com.apple.CoreFoundation      	0x00007fff20480a19 __CFRunLoopRun + 842
22  com.apple.CoreFoundation      	0x00007fff2047ffe2 CFRunLoopRunSpecific + 567
23  com.apple.HIToolbox           	0x00007fff2894e630 RunCurrentEventLoopInMode + 292
[... 454 lines cut ...]
mapped file                        4.1G      400 
shared memory                       64K        4 
unshared pmap                     8544K        5 
===========                     =======  ======= 
TOTAL                              6.4G     1979 
TOTAL, minus reserved VM space     5.5G     1979 

Model: MacBookPro17,1, BootROM 6723.81.1, proc 8:4:4 processors, 16 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: lp_ddr4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Dec 31 2020 21:39:03 version 18.20.222.20.7.8.104 FWID 01-1b2645bc
Bluetooth: Version 8.0.3d9, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 to DVI Adapter
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB 10/100 LAN
USB Device: USB Billboard Device
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

zolrath, 2021-03-07:
Experiencing the same issue here. I've never had SuperSlicer installed on this machine, fresh download and it immediately closes after the splash screen on OS X 11.2.1

ZeekWorld, 2021-03-07:
Same as zolrath.

tinymachines-3d, 2021-03-08:
Crashes after boot logo

Process:               SuperSlicer [2530]
Path:                  /Volumes/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.3.55.5 (SuperSlicer_2.3.55.5)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [2530]
User ID:               501

Date/Time:             2021-03-07 21:20:42.657 -0600
OS Version:            macOS 11.2.2 (20D80)
Report Version:        12
Anonymous UUID:        4F021DE4-B2D2-1F41-409C-5FBCB276D2C1

Sleep/Wake UUID:       640D882F-BE13-46A4-A019-DB995185CF60

Time Awake Since Boot: 8600 seconds
Time Since Wake:       650 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [2530]

VM Regions Near 0x8:
--> 
    __TEXT                      10b2d0000-10d03c000    [ 29.4M] r-x/r-x SM=COW  /Volumes/*/SuperSlicer.app/Contents/MacOS/SuperSlicer

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x000000010ca2194b wxMacCoreGraphicsPenBrushDataBase::CalculateShadingValues(void*, double const*, double*) + 59
1   com.apple.CoreGraphics        	0x00007fff24f641cc CGFunctionEvaluate + 251
2   com.apple.CoreGraphics        	0x00007fff24f6402f function_evaluate + 355
3   com.apple.CoreGraphics        	0x00007fff24f6380e ripc_AcquireFunction + 1028
4   com.apple.CoreGraphics        	0x00007fff24f62a7c ripc_DrawShading + 5645
5   com.apple.CoreGraphics        	0x00007fff252c6c5e CG::DisplayListExecutor::drawShading(CG::DisplayListEntryShading const*) + 636
6   com.apple.CoreGraphics        	0x00007fff25378f4d CG::DisplayList::executeEntries(std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*, bool) + 263
7   com.apple.CoreGraphics        	0x00007fff24ff193b CG::DisplayList::execute(CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*) + 379
8   com.apple.CoreGraphics        	0x00007fff24ff15a8 CGDisplayListDrawInContextDelegate + 529
9   com.apple.QuartzCore          	0x00007fff26c0d0e3 CABackingStoreUpdate_ + 544
10  com.apple.QuartzCore          	0x00007fff26c72141 invocation function for block in CA::Layer::display_() + 53
11  com.apple.QuartzCore          	0x00007fff26c0c4c4 -[CALayer _display] + 2183
12  com.apple.AppKit              	0x00007fff22dddbb1 -[_NSBackingLayer display] + 475
13  com.apple.AppKit              	0x00007fff22d490f5 -[_NSViewBackingLayer display] + 555
14  com.apple.QuartzCore          	0x00007fff26c0b4c8 CA::Layer::display_if_needed(CA::Transaction*) + 874
15  com.apple.QuartzCore          	0x00007fff26d4b5fb CA::Context::commit_transaction(CA::Transaction*, double, double*) + 517
16  com.apple.QuartzCore          	0x00007fff26bece95 CA::Transaction::commit() + 783
17  com.apple.AppKit              	0x00007fff22df2e22 __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
18  com.apple.AppKit              	0x00007fff23541ec2 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
19  com.apple.CoreFoundation      	0x00007fff2044fdad __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
20  com.apple.CoreFoundation      	0x00007fff2044fc3d __CFRunLoopDoObservers + 549
21  com.apple.CoreFoundation      	0x00007fff2044f0ed __CFRunLoopRun + 838
22  com.apple.CoreFoundation      	0x00007fff2044e6ce CFRunLoopRunSpecific + 563
23  com.apple.HIToolbox           	0x00007fff286d6630 RunCurrentEventLoopInMode + 292
24  com.apple.HIToolbox           	0x00007fff286d6282 ReceiveNextEventCommon + 283
25  com.apple.HIToolbox           	0x00007fff286d614f _BlockUntilNextEventMatchingListInModeWithFilter + 64
26  com.apple.AppKit              	0x00007fff22c6e9b1 _DPSNextEvent + 883
27  com.apple.AppKit              	0x00007fff22c6d177 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
28  com.apple.AppKit              	0x00007fff22c5f68a -[NSApplication run] + 586
29  com.merill.slic3r/            	0x000000010ca746ee wxGUIEventLoop::OSXDoRun() + 174
30  com.merill.slic3r/            	0x000000010cbd17c7 wxCFEventLoop::DoRun() + 39
31  com.merill.slic3r/            	0x000000010cb0ae7f wxEventLoopBase::Run() + 143
[... 440 lines cut ...]
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       510.1M        6 
__OBJC_RO                         60.6M        1 
__OBJC_RW                         2449K        2 
__TEXT                           316.1M      332 
__UNICODE                          588K        1 
mapped file                      238.2M       29 
shared memory                       40K        4 
===========                     =======  ======= 
TOTAL                              1.8G     1183 
TOTAL, minus reserved VM space     1.4G     1183 

[... cut ...]
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5444
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1675.1)
Bluetooth: Version 8.0.3d9, 3 services, 19 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0128G, 121.33 GB
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: 2.4G Receiver
Thunderbolt Bus: MacBook Air, Apple Inc., 27.2

cmandlbaur, 2021-03-11:
This looks like a duplicate of #777 which has a solution in comments.

dwisetiyadi, 2021-03-12:
> This looks like a duplicate of #777 which has a solution in comments.

Please read carefully, it is different version (2.3.55.3), my problem is on 2.3.55.5.

I have no issue with previous version.

royal32, 2021-03-12:
> > This looks like a duplicate of #777 which has a solution in comments.
> 
> Please read carefully, it is different version (2.3.55.3), my problem is on 2.3.55.5.
> 
> I have no issue with previous version.

This is indeed the same problem, it showed up in 2.3.55.3, and is still present in 2.3.55.5.
I tagged merill with the solution in the discord almost two weeks ago, but seeing as we're getting numerous issues of mac crashing and the fix hasnt been merged, I'll tag him again on #777

dwisetiyadi, 2021-03-13:
> This is indeed the same problem, it showed up in 2.3.55.3, and is still present in 2.3.55.5.
> I tagged merill with the solution in the discord almost two weeks ago, but seeing as we're getting [...]

2.3.55.33 running fine on my Mac 🤣

closed 2021-06-04 by supermerill