supermerill/SuperSlicer #994: Missing Perimeters on SS vs PS when using 2 instead of 3 vertical Shells

jesterhead82, 2021-03-04:
### Version
2.3.55.5

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
CR10V2 / Marlin 2.x

### Behavior

When I use vertical Shells of 2 instead of 3 get
- Missing perimeters on first layer

Tried different infills, also the nifty quality settings (thin walls, top perimeters), no change. Seems to be the case only for 2 perimeters.

- SS 2.3.55.5 using defaults (Shell 2,0.28):
[image]

- SS 2.3.55.5 (Shell 3,0.28):
[image]

- Works on PS 2.3 using stock settings(2 Shells is default for 0.28):
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[Rear Door Outer Frame.zip](https://github.com/supermerill/SuperSlicer/files/6083789/Rear.Door.Outer.Frame.zip)

supermerill, 2021-03-07:
seems like the one-lopp perimeter isn't correctly written in the gcode anymore

jesterhead82, 2021-03-07:
Hi,

FYI: If I change the one-loop seam to next instead of back its fixed.

commit 515502e6 (2018-07-03): Fixed #994.

Print double-type values according to "full" value instead of 2 digits after point
commit 7b24e836 (2018-07-25): Fixed #994.

Print double-type values according to "full" value instead of 2 digits after point