supermerill/SuperSlicer #995: [2.3.55.5] Broken thin walls on 0.2 Initial Layer Height

jesterhead82, 2021-03-04:
### Version
2.3.55.5

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
CR10V2 / Marlin 2.X

### Behavior
- Unconnected thin walls
- Inconsistent width on thin walls

Only on Initial Layer Height 0.2, works e.g. on 0.32 or 0.16

- Initial Layer Height 0.2 (broken)
[image]

-Initial Layer height 0.32 (works)
[image]

- Initial Layer height: 0.16 (works)
[image]

#### Project File (.3MF) where problem occurs
[Rear Door Outer Frame_0.20InitialLayer.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6083922/Rear.Door.Outer.Frame_0.20InitialLayer.3mf.zip)

supermerill, 2021-03-07:
[image]

jesterhead82, 2021-03-07:
Hi,

it's visible on the two larger parts, right side of the bed, even in your screenshot:

[image]

supermerill, 2021-03-13:
Found it, very hard bug to understand as it's deep inside the boost lib.
As I can't really fix it, I've added a mitigation process: if the lines from boost voronoi are off, then offset the lines a bit and try again.

sttts, 2021-03-29:
First I thought this is the problem I see, but it isn't, though connected to thin walls too:

[image]

As you see with the lower object: perimeter 4 for vertical walls. But the thin wall is hollow, only infill (13%).

The same object with the same profile with 2.2 I get solid walls, as expected:

[image]

sttts, 2021-03-29:
Maybe https://github.com/supermerill/SuperSlicer/issues/991.

closed 2021-06-04 by supermerill