supermerill/SuperSlicer #997: 3mf cannot be opened after a save

robustini, 2021-03-05:
### Version
2.3.55

### Operating system type + version
Windows 10 Pro

### Behavior
I have saved a project and can't open it anymore.
The annoying thing is that once I have established this I have to start saving projects with different revisions each time. 

#### Project File (.3MF) where problem occurs

[Coda_Fusoliera.zip](https://github.com/supermerill/SuperSlicer/files/6088998/Coda_Fusoliera.zip)

supermerill, 2021-03-07:
Your project file has defined a fill apttern for every model, but some are "'empty" (it's in the file SuperSlicer_model.config in metadata inside the zip that is the .3mf)
I tried to replicate, to create a file with that kind of bug but i can't, every time a good name is written in these fields.

What did you do to get this bad save file?

robustini, 2021-03-08:
@supermerill i just saved the file on my nvme, like i do with any other application.
I've never had this type of problem with other programs, so I think the corruption is due to a SuperSlicer issue.

robustini, 2021-03-08:
> Your project file has defined a fill apttern for every model, but some are "'empty" (it's in the file SuperSlicer_model.config in metadata inside the zip that is the .3mf)
> I tried to replicate, to [...]

I tried manually to correct the "fill pattern" in the xml file, since I remember they were all "rectilinear" and now the project file opens without any problem.
But I perfectly remember selecting "rectilinear" when I created the project.

supermerill, 2021-03-13:
Of course, the slicer is the problem, but I don't know why it does this, and so if I want to correct the issue, I have to replicate the problem to understand why it's doing that.

So please try to reproduce the problem, and write here the steps.

robustini, 2021-03-17:
> 
> 
> Of course, the slicer is the problem, but I don't know why it does this, and so if I want to correct the issue, I have to replicate the problem to understand why it's doing that.
> 
> So please try to reproduce the problem, and write here the steps.

Step to reproduce:

1) add modifier box with rectilinear infill to the STL
2) copy and paste this box and move then to another place
3) copy and paste the latest modifier box to another place
4) save

From what I see as long as the "modifier boxes" are only 2 it does not happen, from the third onwards yes.

supermerill, 2021-03-17:
> * add modifier box with rectilinear infill to the STL
> * copy and paste this box and move them to another place
> * copy and paste the latest modifier box to another place
> * save

even with that, I can't reproduce the save bug.
what's your version? (in help-> about)

robustini, 2021-04-26:
@supermerill it is very strange that you cannot replicate the problem.
I just need to add some "modifier" and every now and then copy them, collate them and then move them.
It also happened to me this morning with the version compiled by the current master, and even editing the xml manually now I can not solve the problem.

[Ala_SX_MID.zip](https://github.com/supermerill/SuperSlicer/files/6376365/Ala_SX_MID.zip)

[image]

supermerill, 2021-04-26:
@robustini

Can you use [this test build](https://mega.nz/file/h0R0kZRY#th9MkRSWEg1GJwXY2e6nrbUZgoSIwt2fVZYjnDgj5Lk) (using the superslicer_console.exe)
When you will be exporting a buggy .3mf, some lines should appear in the console, can you write them here?

robustini, 2021-04-27:
@supermerill 

I am trying to use it but everything is terribly slow, is this normal? 
Indeed I would say that at least here it is unusable.
To do the slicing where it used to take 20 seconds now takes over 5 minutes.
During any operation I don't see anything in the console. 

[image]

supermerill, 2021-04-27:
> I am trying to use it but everything is terribly slow, is this normal?

yes, it's debug build

> During any operation I don't see anything in the console.

it's only writing when saving the project, as the problem is here.

robustini, 2021-04-27:
Se the latest lines...

[image]

supermerill, 2021-04-27:
[here is an other test build](https://mega.nz/file/xs5UXLST#CamRwRNnykcIZblSSou6Tz2F9ot5xC1w9XTyfo3iN80) with a possible fix.
So if it creates a problema gain, it will output more logs, and there is a possibility that it doesn't crash.

robustini, 2021-04-27:
Same issue at the first attempt.

[image]

supermerill, 2021-04-27:
Erf, sadely it just confirm that this bugs is very weird.

I cleaned up a bit some definitions that weren't up to the latest c++11 thread-safe standard.
Also added more logs to try to pinpoint exactly what's going on.

[here the build to test](https://mega.nz/file/E8xSGQII#yHBr99NlmHY99CQ51BGf4XSekXZFIYmhLRSrXeN0XzM)

I expect at least 1 or 2 tests to follow.

robustini, 2021-04-28:
Mouse right click on the piece and this appeared.
Obviously then the app closes.
So I tried again to do the same things, created a shape box, rescaled, added some modifier, to the third added it always does this.

[image]

[image]

supermerill, 2021-04-28:
this assert error has nothing to do with our bug, it's only because it's a debug build and some of the gui code isn't that robust.
can you retest some times get back to the usual bug?
If you can't get it back, maybe that means that the bug is resolved .

supermerill, 2021-04-28:
Bad assert, don't let the right selection on a setting, click on the objet/a modifier before right-clic.
note that you can also right-clic on the paper at the right of your object.

>  i see now there's a fix in the Prusa Slicer master

where? can't spot it.

robustini, 2021-04-28:
> 
> 
> Bad assert, don't let the right selection on a setting, click on the objet/a modifier before right-clic.
> note that you can also right-clic on the paper at the right of your object.

You're right!
I added 30 modifiers and it never gave a problem, do you think it's ok now?  
 
> where? can't spot it.

I don't know, but [here](https://github.com/prusa3d/PrusaSlicer/issues/5988) they say that in the 2.4 master the problem has been solved.

supermerill, 2021-04-28:
>I don't know, but here they say that in the 2.4 master the problem has been solved.
yeah!
I'm going to remove the log and let you test it a bit.
Then I'm going to submit the fix also to PS.

> I don't know, but here they say that in the 2.4 master the problem has been solved.

that's for infill, not perimeters.

supermerill, 2021-04-28:
[here is the release candidate}(https://mega.nz/file/5sgikbgA#cfUfDZfxL2fdqiBcz1lIzW1yOTJlklOG7wXAGGi3Mqg)
there are no logs.

robustini, 2022-02-11:
Unfortunately I am sorry to inform you that this is still happening.

[image]

robustini, 2022-03-13:
@supermerill I don't know if it is a similar problem but now it is perhaps worse.
Try opening this saved with today's nightly.

[image]

[FUSO04.zip](https://github.com/supermerill/SuperSlicer/files/8239296/FUSO04.zip)

commit d2e4121b (2021-04-28): fix #997