supermerill/SuperSlicer #998: request horizontal expansion for hole only

tjengbudi, 2021-03-07:
hello,
it really nice to have object printed with exactly the same dimension of my model design. i have calibrated my printer, my object square is calibrated. my filament extruder is calibrated also. but when i desain i block with a hole i cannot plug it nicely even i added 0.1 in my design. i slice it with s3d before and found nothing about horizontal expansion

so i try another slicer. i got prusaslicer, superslicer and cura. i have better printing with super slicer so i ask this. and i have very bad slicing with cura. but cura have hole horizontal expansion so i request this if any option to add this feature.

with xy horizontal expansion my outer dimension is changing also. outer dimension is priority. so i need to not changing the outer dimension. 

please let me know if this option possible to add in the future

thanks a lot

neophyl, 2021-03-07:
This already is there.  Under Print Settings>Slicing
Vertical Hole Shrinking Compensation.  You have a few options, convert to poly holes, or the XY compensation with its associated threshold value.

tjengbudi, 2021-03-07:
oops my bad. thanks for the clarification.

closed 2021-03-07 by tjengbudi

kikomansi, 2022-02-22:
I am confused on how this work. Can someone guide me?

commit 2fae893a (2018-07-03): Fixed #998

Added detection of gtk2/gtk3 GUI libraries required by Alien::wxWidgets.
Added gtk2/gtk3 include paths, so we may call gtk2/3 API directly if needed for some workaround.
commit 287b7723 (2018-07-25): Fixed #998

Added detection of gtk2/gtk3 GUI libraries required by Alien::wxWidgets.
Added gtk2/gtk3 include paths, so we may call gtk2/3 API directly if needed for some workaround.