supermerill/SuperSlicer #999: M104 not issued for Second tool when `ooze_prevention=false`

perrito666, 2021-03-07:
### Version

SuperSlicer 2.3.55.22

### Operating system type + version

Windows 10

### 3D printer brand / version + firmware version (if known)

Geeetech A10 with GT2560v4 board running marlin 2.7.0

### Behavior

When printing with double hotend with `ooze_prevention=false` `T1` is never heated, in my case I see `M104 S205 T0 ; set temperature` appear in the right place but when the turn is for `T1` to print there is no `M104`, this causes extruder to push filament on a cold nozzle.

1. Configure 2 extruders (no shared nozzle)
2. Disable ooze prevention
3. Slice a multi material print
4. Resulting GCode should have only a M104 for T0

I checked latest Slic3r and Prusa Slicer and this behavior is not present in them.

perrito666, 2021-03-17:
I have another data point here, turns out that this only happens if your custom initialization g-code contains a M104, even if that M104 is for the other T

supermerill, 2021-03-18:
yeah, when it detect a M104, it stop doing M104 itself and let you do your stuff without meddling with you

perrito666, 2021-03-18:
Would it be possible to issue a warning when resulting gcode never heats T1 even if E1 is in fact used? I see very likely that people would just issue M104 for the first Hotend to prime it before beginning the print (as I was doing by accident, M104 without T param only caused selected tool to be heated) and have that turn off T1 for the whole print. (notice that this even omitted the temperature changes T1 should have done)

supermerill, 2021-03-18:
Maybe but not in the short term.

closed 2021-06-14 by supermerill