supermerill/SuperSlicer #1000: Request: Exclude area of build surface

ProfEngr, 2021-03-08:
I just discovered that my cooling duct will impact the strain relief clip for the bed heater wires in the volume bordered by the corners 0,225,0 and 45,235,10. I know I can just write myself a note to keep things out of this area, but I can only eyeball paths so well. Is there, or could there be, a way to tell the slicer to place that corner out of bounds as it would with items larger than the normal build volume?
Thankfully I noticed the interference prior to starting the print.

supermerill, 2021-03-18:
You can create your own bed model.
create an stl (with your preferred tool or the included script engine)
example:
```
cut()(
cube(250,250,1),
cube(10,10,1).move(0,45),
cube(10,10,1).move(0,235),
)
```
Then import it in bed shape -> form personalised -> load stl.

supermerill, 2021-03-19:
also duplicate of #176

ProfEngr, 2021-03-31:
Had to change the coordinates of your example, but I got the shape generated. Saved as an STL. Went into Printer and loaded it as the bed model, then I got this as the result.

[image]

supermerill, 2021-03-31:
do not load it as bed model, but as a custom bed shape 
[image]

ProfEngr, 2021-03-31:
Oops. Saw that this time around. It shows the notch correctly, but I can still place an object in that notch w/o it turning out-of-bounds blue. It does give me a visual reminder that the area should be off limits at least. Should it turn blue though?

[image]

supermerill, 2021-03-31:
seems like it's using the AABB bounding box of the plater for  the out-of-bounds computation. Not that great as a feature, as so it's not different than using a custom model...

jscancella, 2022-10-17:
@supermerill  Any chance you will allow a hole in the future? I have a defect on my build plate that is exactly in the middle and I often forget to move things around so that defect doesn't show up on my printed items.