supermerill/SuperSlicer #1003: Error opening cooling section of filaments

bigsubgit, 2021-03-11:
### Version
2.3.55.5_

### Operating system type + version
Windows 10 Pro (20H2 version)

### 3D printer brand / version + firmware version (if known)
Custom Ender3 with klipper

### Behavior
I've left macos Catalina and install a new windows 10 machine. I esported all profile from 2.3.55 on macos and reimported them in windows.
Now when I open cooling section of filament, Superslicer crashs with error attached.

[image]

[image]

This is an extract of config bundle, about one filament:

`[filament:E3K_Basicfil PETG Orange]
bed_temperature = 70
bridge_fan_speed = 60
chamber_temperature = 0
compatible_printers = "0.4 nozzle Ender3 Klipper"
compatible_printers_condition = 
compatible_prints = 
compatible_prints_condition = 
cooling = 1
disable_fan_first_layers = 3
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
external_perimeter_fan_speed = 0
extrusion_multiplier = 1
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #FF8040
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
filament_cooling_zone_pause = 0
filament_cost = 19.95
filament_density = 1.34
filament_deretract_speed = nil
filament_diameter = 1.72
filament_dip_extraction_speed = 70
filament_dip_insertion_speed = 33
filament_enable_toolchange_part_fan = 0
filament_enable_toolchange_temp = 0
filament_load_time = 0
filament_loading_speed = 28
filament_loading_speed_start = 3
filament_max_speed = 0
filament_max_volumetric_speed = 0
filament_max_wipe_tower_speed = 0
filament_melt_zone_pause = 0
filament_minimal_purge_on_wipe_tower = 15
filament_notes = ""
filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
filament_retract_before_travel = nil
filament_retract_before_wipe = nil
filament_retract_layer_change = nil
filament_retract_length = nil
filament_retract_lift = nil
filament_retract_lift_above = nil
filament_retract_lift_below = nil
filament_retract_restart_extra = nil
filament_retract_speed = nil
filament_settings_id = ""
filament_shrink = 100%
filament_skinnydip_distance = 31
filament_soluble = 0
filament_spool_weight = 0
filament_toolchange_delay = 0
filament_toolchange_part_fan_speed = 50
filament_toolchange_temp = 200
filament_type = PET
filament_unload_time = 0
filament_unloading_speed = 90
filament_unloading_speed_start = 100
filament_use_fast_skinnydip = 0
filament_use_skinnydip = 0
filament_vendor = (Unknown)
filament_wipe = nil
filament_wipe_advanced_pigment = 0.5
filament_wipe_extra_perimeter = nil
first_layer_bed_temperature = 75
first_layer_temperature = 225
full_fan_speed_layer = 0
inherits = 
max_fan_speed = 35
max_speed_reduction = 90%
min_fan_speed = 35
min_print_speed = 15
slowdown_below_layer_time = 5
start_filament_gcode = "; Filament gcode\n"
temperature = 220
top_fan_speed = 65`
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

bigsubgit, 2021-03-11:
EDIT: Just tried to switch from italian to english language and the error is gone away.

supermerill, 2021-03-13:
same as #1004

bigsubgit, 2021-03-13:
> same as #1004

Ok thanks

closed 2021-08-22 by supermerill