supermerill/SuperSlicer #1005: Draft shield doesn't works if complete objects mode is ON

ZhenyaKa, 2021-03-12:
### Version
2.3.55

### Operating system type + version
Win 10

Can't print smal sequintal objects with draft shield - it doesn't slice any shield in that case

neophyl, 2021-03-12:
If you want a work around to tide you over you can turn off the draft shield option but set the skirt height to something very large, like 1000 layers.  Draft shield is just a auto height skirt anyway.  When you do that it slices a shield around each part even when sliced individually.

ZhenyaKa, 2021-03-13:
> If you want a work around to tide you over you can turn off the draft shield option but set the skirt height to something very large, like 1000 layers. Draft shield is just a auto height skirt anyway. When you do that it slices a shield around each part even when sliced individually.

I can model the shield too :) But why slicer doesn't do that it must to do? :)

neophyl, 2021-03-13:
I'm just a user like you but from memory Merrill made quite a lot of changes on the previous versions based on PS 2.2 to do with individual prints and I think he added the individual draft shield option too which PS doesnt have.  

These latest versions are based on PS2.3 and Prusa made many many changes to the way it works under the hood, which Merrill now has to go back through and redo all the changes he made to make them work with 2.3 now.  That is a lot of work.  
None of the current 2.3.55 releases are proper 'stable' releases, which is why they are still using the alpha locations for configuration storage.  
Basically if you are using 2.3.55.x then you are using alpha/beta software which is bound to have bugs in at the moment.  Just carry on reporting them here so that Merrill knows about them and wait until he can fix them.

Mean while if you do run into an issue then either use a work around or run the 2.2.x.x stable release.

closed 2021-06-04 by supermerill