supermerill/SuperSlicer #1010: Variable Layer Height doesn't always work

mrkoz, 2021-03-14:
### Version 2.2.53
### Operating system Windows 10 - all latest service packs

### 3D printer : fully custom built CoreXY of my own design, running Marlin 2

### Behavior
* in some cases, it wont take into account the variable layer height until i resize the part.

### Steps:
1. Generate a 40mm voron test cube
2. set variable layer height on with defaults
3. click adaptive once
4. click slice preview
5.  -- layers not varied --
6. back out of preview and resize the voron cube to 99.9%
7. set varliable layer height on with defaults
8. click adaptive once
9. click slice 
10.  -- layers varied --

pre-slice at 100%
[image]

sliced at 100% with dynamic layers
[image]

resized cube to 99.9% then re-ran dynamic layers
[image]

Project file:
[voron_design_cube_v6.zip](https://github.com/supermerill/SuperSlicer/files/6136235/voron_design_cube_v6.zip)

Pavulon87, 2021-03-15:
It's a bit weird, but I can confirm it.
However, this is not a problem for the 40mm cube(box shape).

mrkoz, 2021-03-15:
> 
> 
> It's a bit weird, but I can confirm it.
> However, this is not a problem for the 40mm cube(box shape).

I have had this on a few other files, even a resize of 0.01% corrects it.

TheAndr0id, 2021-05-02:
> > It's a bit weird, but I can confirm it.
> > However, this is not a problem for the 40mm cube(box shape).
> 
> I have had this on a few other files, even a resize of 0.01% corrects it.

I can confirm this as well.  I have something that needs to be printed at a specific height that is not a multiple of layer height.  I looked at variable layer height to use the top few layers to get closer to the height needed and it did nothing.  I scaled the item to 100.01 and now the variable layer height works.  This is a flat item (coin) with little to no surface detail.

I am using version 2.2.53.4 under Linux x64 (build from source).

supermerill, 2021-05-03:
an dyou have the `printer->general -> enable variable layer height feature` enabled?

supermerill, 2021-05-03:
Can't replicate (win10 also), with latest nightly

i opened your porject file
clic on varaible alyers,
clic on adaptative
clic on slice
=> i have some thin layers.

TheAndr0id, 2021-05-03:
Thank you @supermerill for looking into this, I'm looking forward to the next release!

mrkoz, 2021-05-03:
> 
> 
> and you have the settgin `printer->general -> enable variable layer height feature` enabled?

Yes, it works as expected 99% of the time, just for some files (potentially height based) it doesn't do the variable layer height part.

Ps @supermerill thank you for all your work, this is a brilliant fork!

closed 2021-06-04 by supermerill

navr32, 2021-11-07:
Hello !
Excuse me I have update to 2.3.57.4 and the variable layer height is always Grey ??? i am unable to activate it again ?
I have try scale or others but nothing.
I am on linux manjaro so Arch distrib clone..
Someone have any idea or how troubleshoot ? many thanks.
Have a nice days.

navr32, 2021-11-08:
Oh excuse me for this bad post... So i haven't see the option in "Printer settings"  "General" 
And the last line on bottom "Enable variable layer height feature" was disable..so just check it and it's ok..all works again.
Many thanks again to all enthusiastic dev and others.