supermerill/SuperSlicer #1011: 1 Vertical shell issue,

Vybramo, 2021-03-14:
### Version
SuperSlicer`_ 2.3.55.5

### Operating system type + version
Windows 10 pro

### 3D printer brand / version + firmware version (if known)
Ender 5 , direct drive , skr 1.4 turbo , Marlin 2.7.2

### Behavior
Hello,
I have an issue with printing wing with super slicer 1 wall because as we can see on photo its go trough wall. Simplify 3D dont have this problem. I try to set all settings possible combination but issue still its same. 
I test in all version which one was released in year of Super slicer, Prusa Slicer and result its same. Settings are : 0 Infill, Perimeter 1, Solid layer TOP/Bottom 0

[image]

[image]

#### Project File (.3MF) where problem occurs

[test.zip](https://github.com/supermerill/SuperSlicer/files/6136252/test.zip)

neophyl, 2021-03-14:
If you do a search on the GitHub for Prusa slicer or even the Prusa forum you will see lots about single perimeter plane wings. Most of the models are not valid constructs in terms of slicing and the s3d ones use some specific tricks of the slice engine to work. 
it’s gotten quite heated at times on the Prusa GitHub thread. I suggest you read through it to get a background to this issue. 

I’ll try and find the direct link when Im back at my pc. Searching for stuff on a phone sucks.

supermerill, 2021-03-17:
the way the part is made: having a little void inside of it and using "overlapping perimeters" => impossible to fix.

But if you use a 0.3 nozzle with a 0.35 external perimeter width, you should be able remove the defects.

closed 2021-06-04 by supermerill