supermerill/SuperSlicer #1012: Perimeter offset

Jack77777777, 2021-03-14:
See attached pic of a simple box. I have a "perimeter offset" at height of top layer from the ground of model. Could this be caused by "only one perimeter on top layer", too much cooling, monotonic too infill or is it a bug. 

I am using SS 2.3.55.5 on Win10 64bit. 

[image]

supermerill, 2021-03-17:
most probably, it's because you have a long printing time at this layer, so it has much more time to cool.

Can't say much more wthout the project file.

Jack77777777, 2021-03-19:
I had auto cooling enabled and it could be that the fan speed up for the above layers due to fan_below_layer_time 60s. Testing with same speed for all layers now.

Jack77777777, 2021-03-19:
Seems my isues are fixed. Thanks @supermerill

closed 2021-03-20 by supermerill