supermerill/SuperSlicer #1014: Bridge Infill - not getting the cooling it deserves?

thierryzoller, 2021-03-15:
### Version
2.3.55

### Operating system type + version
Windows

In my object there the majority of bridges get the right fanspeed, however there are at least two, that's don't. Is this expected? It's a 90° hanging object so I'd kind of expect the increased fan speed.

[image]

thierryzoller, 2021-03-15:
[Form Giesen Lautsprecher.zip](https://github.com/supermerill/SuperSlicer/files/6142872/Form.Giesen.Lautsprecher.zip)

The-EG, 2021-03-16:
I've been seeing this same issue. It appears that 'bridge infill' isn't getting the 'bridges fan speed' setting applied if they have perimeters on all sides or something.

Pretty easy to reproduce:
1. Add shape-> Cube (20x20x20)
2. Slice it with a fan settings of 50% default speed, 50% max fan speed, 100% bridges speed.
3. Observe the first top solid layer has 50% fan speed, not 100%:
[image]
[image]

Not the case in PrusaSlicer (2.3, eef800b5c4215abc9226cebae5733227190d9fd8):
[image]

supermerill, 2021-03-17:
I think it's a "fan speed-up" bug. It will be rewrote on the next makjor version.
Deactivate the option, in printer-general ( set it to 1)

thierryzoller, 2021-03-21:
Setting it to 1 did not change the problem

supermerill, 2021-03-21:
sorry, it's a typo, it's 0 to deactivate it.

The-EG, 2021-05-29:
This appears to be working in master now.

closed 2021-06-04 by supermerill

thierryzoller, 2021-07-11:
Still seem to have the same (similar issue) with the latest build 2.3.56.7. Bridge infill gets default 70% instead of 100%

[image]

thierryzoller, 2021-07-11:
Was Fan delay of 0.5s. My question then is : is why doesn't it start 0.5seconds **_earlier_** rather than later ? It seems to not speed up with 0.5s delay

[image]

supermerill, 2021-07-11:
It's buggy, i'm rewriting the gcode writer, and after that it will be easier to make something relibale.