supermerill/SuperSlicer #1015: Paint on seam not working as expected

ogland, 2021-03-16:
### Version
Version 2.3.55

### Operating system type + version
Windows, version 10

### 3D printer brand / version + firmware version (if known)
Klipper

### Behavior
Seam not ending up where painted on seam
Paint seam in inner corners, slice
Expecting seam to end up where it's painted
[image]

Not ending up where it's painted
[image]

#### Project File (.3MF) where problem occurs
Can't share this model

ogland, 2021-03-16:
Ok so on closer inspection, they are indeed where they are supposed to be, but not visible through.
[image]

thierryzoller, 2021-03-16:
Seems like it works?

closed 2021-03-17 by supermerill