supermerill/SuperSlicer #1016: Multimaterial Toolchange Temperature cannot be set to a value below 175°C

well-engineered, 2021-03-16:
### Version
2.3.55

### Operating system type + version
osX

### 3D printer brand / version + firmware version (if known)
E3D Toolchanger

### Behavior
Setting: Filament -> Multimaterial Toolchange Temperature -> toolchange temperature -> inputfield

The parameter can only be set to a value between 175°C and 285°C. 

Please let the user adjust this value fully.

Additionally: The G10 commands for this option are put at the very first lines in the generated .gcode file. Even before all the usual comment-lines. I expect them to be somewhere near the place where the settings for the bed-temps are put.

well-engineered, 2021-03-17:
Well, that was quick! Thank you :)

closed 2021-06-04 by supermerill

24c, 2021-11-20:
This is not the same issue, but using 2.3.57.6 on macOS I've noticed when "Toolchange temperature enabled" is selected, the M104 T1 temperature is increased by 20ºC in my case from 240 to 260ºC  in the exported gcode. I haven't got any values like this anywhere, so unsure how this is calculated. The project file is just a two colour traffic cone, that I am trying to use on a Climber7 IDEX printer with ooze wiper arms fitted.
[M104_S260_T1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7575146/M104_S260_T1.3mf.zip)

[image]
.

supermerill, 2021-11-20:
@24c 
you set 20°C as temp offset in your second extruder...

24c, 2021-11-20:
I'm sorry, I inherited this profile, and never saw it. Thanks for putting me right. :doh