supermerill/SuperSlicer #1018: Areas of Support starting in mid air

CCS86, 2021-03-17:
2.3.55.5 / Win10 x64

[image]

[Support-mid-air.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6153298/Support-mid-air.3mf.txt)

neophyl, 2021-03-17:
Exists in PS too.  Common issue.

supermerill, 2021-03-17:
isn't it more of a "support start on top of a previous object layer but as there is a wide gap, it's irl in the air?

CCS86, 2021-03-17:
It is definitely starting in mid air. I tried to open the project file to get more screenshots, but get an error:

[image]

supermerill, 2021-03-18:
weird, i have no problem loading it.

CCS86, 2021-03-18:
Interesting. If I double-click the 3mf file it fails with that error. 

If I choose open from within SS, it works fine.

supermerill, 2021-03-18:
If you have prusaslicer as default opener for .3mf, i just saw that i forgot to translate "monotonicgapfill" to something it understand.