supermerill/SuperSlicer #1021: [Linux] "Print host upload queue" opens a new independent window but is not visible on task bar

JohanAR, 2021-03-19:
### Version
2.3.55.5

### Operating system type + version
Xubuntu 20.04

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* Select "Print host upload queue" from menu
* New window opens (on primary monitor, even if SS is on a different monitor)
* The new window does not get its own box on the task bar

This is a problem for me because during the day I use my other monitor (which according to the OS is my primary) for work, and without the task bar entry it is impossible for me to move the upload queue window to the monitor where I can see it.

#### Project File (.3MF) where problem occurs
N/A

supermerill, 2021-03-19:
weird. I have the dialog box in my task bar on windows, and I can't see any os-related code that may change the behavior. I'm wondering if it's a limitation of the linux  wxwidget version, or a setting of your window manager that create the issue.

JohanAR, 2021-03-19:
Been trying to figure out if there's some sort of default behaviour for this in Linux, so I tried opening child windows in other programs. "Yes" means that the child window appears on the task bar.

Thunar prefs: Yes
Thunar about: No
GEdit prefs: No
GEdit about: No
Firefox prefs: N/A (uses tab instead of new window)
Firefox about: Yes
GIMP prefs: Yes
GIMP about: Yes
Blender about: N/A (not a window)
Blender prefs: Yes

Thunar, GEdit and Firefox can open new windows, but these are identical to the main window so I think it's obvious that they would get a new task bar entry (like opening several instances of the same program).

GIMP can break out tool panels to separate windows, these do not appear on the task bar. While the tool windows are movable to different screens, they appear to be sort of attached to the main window, and if I click the main window it brings all the other windows to front as well.

Blender can open both more main windows, and customizable non-main windows which appear on the task bar.

So I think there is no standard behaviour, but personally I think it's better if SuperSlicer's extra windows get task bar buttons.

I'm trying to dig a little into the code to see if I can figure out if there is some setting with different default behaviour on Windows/Linux

JohanAR, 2021-03-19:
The problem seems to be that it inherits DPIDialog rather than DPIFrame. I would guess Linux desktop environments, or at least Xfce that I'm using, doesn't show dialogs on the task bar, but perhaps Windows doesn't have different types of windows in the same way

supermerill, 2021-03-20:
Sorry but I don't have time to code & try the gui for every Linux distribution.
If you or someone can come with an elegant code solution that won't change the intended behavior in other os & won't break other Linux distribution, I'll happily merge it.

JohanAR, 2021-03-20:
I can definitely do it. I did some experiments yesterday converting this window to a DPIFrame, and it looked promising. It requires a few small tweaks in addition to changing the base class, but it will be pretty clean. Now the questions are:

1) Should I change some other sub-windows too? A brief search through the code indicates that it's only the main window that inherits DPIFrame, while everything else is based off DPIDialog. I think there might not always be a clear line between what is a dialog and what is a frame, but imho some of these ought to be reclassified as frames.

```
~/P/SuperSlicer ❯❯❯ g grep "public DPIDialog"                                                                         master 1✭ 7✱ 2◼
src/slic3r/GUI/AboutDialog.hpp:class CopyrightsDialog : public DPIDialog
src/slic3r/GUI/AboutDialog.hpp:class AboutDialog : public DPIDialog
src/slic3r/GUI/BedShapeDialog.hpp:class BedShapeDialog : public DPIDialog
src/slic3r/GUI/CalibrationAbstractDialog.hpp:class CalibrationAbstractDialog : public DPIDialog
src/slic3r/GUI/ConfigSnapshotDialog.hpp:class ConfigSnapshotDialog : public DPIDialog
src/slic3r/GUI/ConfigWizard.hpp:class ConfigWizard: public DPIDialog
src/slic3r/GUI/ExtruderSequenceDialog.hpp:class ExtruderSequenceDialog: public DPIDialog
src/slic3r/GUI/FreeCADDialog.hpp:class FreeCADDialog : public DPIDialog
src/slic3r/GUI/KBShortcutsDialog.hpp:class KBShortcutsDialog : public DPIDialog
src/slic3r/GUI/PhysicalPrinterDialog.hpp:class PhysicalPrinterDialog : public DPIDialog
src/slic3r/GUI/Plater.cpp:class ProjectDropDialog : public DPIDialog
src/slic3r/GUI/Plater.cpp:class ProjectDropDialog : public DPIDialog
src/slic3r/GUI/Preferences.hpp:class PreferencesDialog : public DPIDialog
src/slic3r/GUI/SavePresetDialog.hpp:class SavePresetDialog : public DPIDialog
src/slic3r/GUI/SysInfoDialog.hpp:class SysInfoDialog : public DPIDialog
src/slic3r/GUI/UnsavedChangesDialog.hpp:class UnsavedChangesDialog : public DPIDialog
```

Those are the classes in question, PrintHostQueueDialog doesn't show up because I've changed it already

2) If I change the base class to DPIFrame, should I also change the name of the class (and in some cases the file name) accordingly? Personally I'm not a fan of misleading class/variable names, but I'm aware that changing these would make it more difficult to merge changes from PrusaSlicer

supermerill, 2021-03-20:
>  make it more difficult to merge changes from PrusaSlicer

Try to minimize the number of modified lines.

> Should I change some other sub-windows too? 

Do they have the same problem when you use them or not?

JohanAR, 2021-03-21:
> Do they have the same problem when you use them or not?
 
Yes, all sub-windows have been classified as wxDialog by their implementors, and it's my opinion that this is incorrect and some of them ought to be changed to inherit wxFrame instead. My theory is that on Windows this distinction might not exist on the OS level, while on Linux a dialog is not given a task bar entry.

This is what wxWidgets say about dialogs:
> Dialogs usually contain either a single button allowing to close the dialog or two buttons, one accepting the changes and the other one discarding them (such button, if present, is automatically activated if the user presses the "Esc" key).

It doesn't describe frames as much, but I suppose a frame would be any top level window which isn't a dialog :) The problem is that I don't think it's always so easy to say which category a windows belongs too. I think one could try to answer the question "would it be useful to move this window to a secondary screen, and keep it open while I continue working in the main window?", in which case it ought to be a wxFrame.

On the other, dialogs would be windows that interrupt your workflow to present some information, or ask the user a question, and are soon closed again. They don't necessary have to be modal and prevent the user from interacting with the main window, but it would make less sense to do so from a UX perspective. For example if the user presses the "Add..." button you get a file selector dialog, and it would be pretty strange to move it out of the way and keep it open while continuing working with the plater. I think it's an obvious example and others will be less clear cut, but I'm just trying to explain what I mean :)

Following this reasoning I think at least the print queue window and the settings window (used by the multi-window layout mode) should be made to wxFrames. And in addition I think all the dialogs should be changed to open centered on their parent window (or possibly on the monitor where your mouse cursor is). Maybe this is also different on Windows, but for me all the wxDialogs open on the primary monitor instead of the one where I am currently working. OS dialogs like the file selector opens on top of the main slicer window like I would expect. This is however just my opinion, and I'm open to counter-suggestions