supermerill/SuperSlicer #1022: [Feature Request] Adjust Infill-Perimeter Overlap for top layer only

Makhaira, 2021-03-20:
At present there are already several modifiers to allow top-layer fine tuning:
- One perimeter only on top layer
- Top layer extrusion width
- Top layer infill flow

By using these in combination you can get an excellent top layer finish, however it often leaves gaps between infill/perimeter due to the reduced flow. 

The ability to adjust infill/perimeter overlap for just the top layer would avoid causing defects in the lower layer of the prints, whilst ensuring the best possible top layer.

supermerill, 2021-03-20:
I think i have to create a "global modifier for top surface" section, to avoid many requests like these.

Makhaira, 2021-03-20:
That would be excellent!

pranas, 2021-11-05:
> I think i have to create a "global modifier for top surface" section, to avoid many requests like these.

You should do the same for the 1st layer as well. You typically want a lot of control for that one as well. By the way, thanks, for adding only 1 perimeter option to 1st layer 👍 Maybe these global modifiers could use similar UI to object modifiers where you can select and add whatever option you want. It would not clutter the UI with tons of fields and height-based modifiers already exist at per-object level.

supermerill, 2021-11-05:
> You should do the same for the 1st layer as well

sure #1339

#1577