supermerill/SuperSlicer #1023: [Feature Request] Ability to generate multiple parts with varying parameters

Makhaira, 2021-03-20:
This is similar to what is done by the in-built flow tuning wizard.

As you get into the late stages of tuning, it is common to need to test multiple iterations of a few parameters. As an example, you might want to test all the top flow widths from 0.3mm to 0.4mm whilst also varying top flow % from 80% to 90%.

At present this involves making multiple copies of each part and repeatedly changing the parameters - it can take quite a while to set up!

The ability to either:
    a) Tell SS what grid size of items to print, and what parameters to vary in X and Y
    or
    b) Pick an item with a modifier (let's say number of perimeters and flow %) and then be able to paste an element (let's say flow%) to a number of other parts with the click of a mouse. 
would save a lot of time and make SS incredibly powerful to set up plates of tuning parts, something that as far as I know no other slicer has ever done.

supermerill, 2021-03-20:
why not, But I don't have the time to code that.

Makhaira, 2021-03-20:
Hmm. I know very little on coding, but where in the files is the current
code to generate the flow pieces please merill? I might see if I can figure
out how to modify it.

supermerill, 2021-03-21:
you can look at src/slic3r/calibrationBridgeDialog.cpp
it contains all the important things.
You can create an other calibration dialog, with a text field & button to a fileDialog to get the stl to duplicate.

you'll have to find a way to let the user input the setting(s) to modify. You can use a(some) simple text entry, for the setting key & value, the api can accept that, if well formatted.

mneuhaus, 2022-02-17:
couldn't the FreeCad python Script integration be extended by some wrapper methods, that control the slicing settings for separate objects?

That way someone could create a script to generate various calibration prints with varying settings.

mneuhaus, 2022-02-17:
hmm, ok that's a pity.
Using a Python API Wrapper would enable pretty solid and flexible creation of such a functionality.
There isn't any other python scripting tied to api anywhere in superslicer so far, right?
I'm pretty fluent in php, python, etc. but never really got a hang with c, etc. 
If there is some more ground work i could build on i might be able to get something going

supermerill, 2022-02-17:
There is no python inside SuSi.
The freecad script use the freecad python.