supermerill/SuperSlicer #1030: Preset bundle `Creality` version not voind in index: 0.1.7

cbc02009, 2021-03-23:
### Version
2.3.55.5

### Operating system type + version
Windows 10 Pro 20H2

### 3D printer brand / version + firmware version (if known)
Voron 2.4 and Creality Ender3 pro, both running latest version of Klipper

### Behavior
On program startup I always get the following error:
[image]

Everything works fine, but nothing I do stops the message from popping up again at startup.

This started after I added my ender 3 to super slicer.

Definitely not a high priority issue since everything works fine, but it would be nice if it would go away once and for all.

supermerill, 2021-03-23:
I think it's quirk that comes from a change of version numbering.
You have to delete the creality.idx that is on your setting directory.
should be in C:\Users\myname\AppData\Roaming\SuperSlicer\cache

cbc02009, 2021-03-23:
> 
> 
> I think it's a quirk that comes from a change of version numbering.
> You have to delete the creality.idx that is on your setting directory.
> should be in C:\Users\myname\AppData\Roaming\SuperSlicer\cache

That doesn't seem to have fixed it. The file was in SuperSlicer-alpha, if that makes any difference.

supermerill, 2021-03-27:
mmm... maybe it contains also an old creality.ini in vendor. Can you replace it by the one that is in superslicer/resources/profiles?

cbc02009, 2021-03-28:
That fixed it. Thanks!

closed 2021-03-28 by cbc02009