supermerill/SuperSlicer #1032: Feature request: dynamic extra length on layer restart

robustini, 2021-03-24:
Printing airfoils with a very low infill and with the extrusion flow calibrated micrometrically to have the minimum weight and a good resistance I've this big problem.
As you can see from the image at the layer change when it starts to extrude it lacks material, but it does not always do so, only when the travel of the nozzle is long enough.
At that moment the nozzle empties itself, necessarily printing PLA at about 230°, which is imperative temperature this type of printing.
So by repeating the printout the result is the same, always missing in the same points.
It does not happen where the travel of the nozzle is low, that is when it has to make little movement to restart with the new layer.
Unfortunately "aligned" works relatively, often the nozzle travels from the leading edge to the trailing edge of the wing at the end of the layer, thus creating this problem.
It could be solved by inserting an "extra length on restart" value, but this would also affect where the problem does not arise, thus creating an accumulation of material.
It would therefore be interesting to have a "dynamic extra length", calculated during the slicing over a nozzle travel distance, and which is obviously modular.
Example: if I set "dynamic extra lengh" to 1mm on 100m of travel, if the movement before the restart of the nozzle will be 50mm it will have an extra length of 0.5mm, therefore the value should dynamically adapt according to the length of the travel.

[image]

burner-, 2021-03-26:
Do you youse Linear Advance at printer firmware? This problem sounds that you do not use it or it is not calibrated right.

robustini, 2021-03-26:
> 
> 
> Do you use Linear Advance at printer firmware? This problem sounds that you do not use it or it is not calibrated right.

No, I use Klipper and for this type of printing the "pressure advance" (similar to Marlin's Linear Advance) is disabled.
I did other tests and playing with the extra restart distance I reached an acceptable compromise, but the feature I ask for would certainly give a better result.

[image]

burner-, 2021-03-26:
Why it is disabled? Pressure advance is exactly solution this and many other this kind problems. When it is calibrated right it make retract need to marginal and specially if your hw is ok then extra restart leght should not be needed at all with PA/LA. Set your retract and extra extrusion to 0 and calibrate PA. Then calibrate retract if needed and only after that add extra extrusion. But if it is needed check that your bowden pipe does not have slack at connection.  Yes in theory this is possible to workaround with dynamic restart lenght but then you just try fix symptoms not cause. Also keep mind that this dripped material what you compensate with extra had to go somewhere (probably for of plobs or over extrude) so it is also reason why it is better to fix that problem it self instead of compensate several different symptoms one by one.

robustini, 2021-03-27:
The Pressure Advance does not solve this problem, it serves another purpose, since it does not dynamically act on the amount of movement of the nozzle.
As you can see from my prints it is well configured, the seams are practically invisible.
This type of printing, on the other hand, creates other situations, which cannot be solved by the Pressure Advance since that too is not dynamic.

[image]

[image]

burner-, 2021-03-27:
"I use Klipper and for this type of printing the "pressure advance" (similar to Marlin's Linear Advance) is disabled." - "As you can see from my 3DBenchy prints the PA it is well configured" So is it configured or not? Yes PA is mainly ment to fix different things. Anyhow extruder is many times leaky because of remaining pressure. PA & retract should solve that. If your nozzle leak so much that you need extra lenght at start first question is that why it is so? Is your travels too slow. And yes dynamic extra lenght will resolve that you do not get holes any more. But it not resolve that you still have extra material at somewhere else. I am not against of this feature but I just try say that you try resolve this problem from wrong place.

robustini, 2021-03-27:
> 
> 
> But it not resolve that you still have extra material at somewhere else.

I agree about this, but whether this extra material remains within the perimeters is irrelevant.
I repeat, PA is disable only for this slicing profile, I reset it in the g-code start, but it is normally active for other prints:

;SET_PRESSURE_ADVANCE ADVANCE=0 SMOOTH_TIME=0

supermerill, 2021-04-03:
You have a big oozing problem and need a more precise feature than the current "extra length on unretract" ?

robustini, 2021-04-04:
> 
> 
> You have a big oozing problem and need a more precise feature than the current "extra length on unretract" ?

@supermerill the oozing problem is related as I wrote above to the need to print certain things at very high temperatures, normally I don't have this problem obviously.

robustini, 2022-04-05:
@supermerill we have performed many cross tests with different printers and it is obviously not a problem of the slicer, but a need to be filled.
In this type of printing to a single wall with a very bland infill **and overtemperature** it often happens that it finishes making the infill in the opposite position to where it will have to start with the external wall and here is missing material because it has exuded it internally as it moves.
Even doing the outer perimeter after the infill the problem comes up.
In my opinion the system that I have proposed would work, it would be enough to find the correct value of the "dynamic" extra restart distance obviously calculated on the maximum displacement based on the size of the bed, so much the material thrown in that runs there will always be in this type of printing.

robustini, 2022-04-05:
Pictures sent to me by a friend, he too problem.
Obviously putting two external walls the problem would remain internal and would not be visible, but we cannot do it for a matter of weight.

[image]

[image]

tonysmartantzoulatos, 2022-04-05:
@supermerill i would say that @robustini has a point with this request. Something like this would drammatically increase print quality on such specialty one perimeter thin prints that need to have there retraction on such a tight spot as the trailing edge. I have seend this happen: Quote: In this type of printing to a single wall with a very bland infill and overtemperature it often happens that it finishes making the infill in the opposite position to where it will have to start with the external wall and here is missing material because it has exuded it internally as it moves.
and while you can mediate it with extra restart, different PA you can never have two print looking exactly the same because it all depends on the end of infill and the ditance it travels, even it it oozes an extra 0.05mm then you can see it on the print and if it starts happening it can go a long distance (Z height) eventually seeing what it seems like an underextrusion issue. I think the idea of the feature is very interesting to implement
[image]

[image]
As you can see on picture one the print is perfect and that next is the print right after that and the issue is seen right after the retraction.

tonysmartantzoulatos, 2022-04-05:
In addition to my previous comment, I decided as proof of concept to run the same gcode once more. Completely different results.
[image]
[image]
Same file same filament, same settings as the perfect print, I am ruling out mechanical issues since you can give me any file to print with a higher perimeter/infill count and it will be flawless. But the result is different

robustini, 2022-04-07:
> In addition to my previous comment, I decided as proof of concept to run the same gcode once more. Completely different results

Yes, it depends on how much the nozzle exudes when moving to start the perimeter again.
It might help to mount a 0.3 nozzle but always leaving the walls at 0.4 surely it would be more difficult to release unwanted material, but I think that with the feature I asked for the problem would be solved, or in any case masked in order to make it irrelevant.

supermerill, 2022-04-07:
"dynamic extra length on restart" may create big seams.
I'm wondering if I can't "prime" the nozzle a bit on the inside/infill area just prior to start the perimeter.

robustini, 2022-04-07:
> I'm wondering if I can't "prime" the nozzle a bit on the inside/infill area just prior to start the perimeter.

Of course that might work too!
Create a prime attached to the perimeter internally, as for two perimeters but one is just a peak on the inside for purging.
But in that way there would be more material in the print.

> "dynamic extra length on restart" may create big seams.

No, if in fact it is dynamic.
The friend with those problems has now printed with the "nearest" seam, i can see the seam but no problem of lack of material, almost pefect.
If the extra restart is dynamic and the displacement is minimal it will not act heavily, so the seam will appear identical.
For example if the dynamic is set with an extra restart distance of 0.1mm per 100mm of displacement, if it is moved by 10mm the extra will be 0.01mm, even if there is not rilevant.
For me it would work, and there would be no extra material like with the prime perimeter.

robustini, 2022-04-14:
As suggested by @tonysmartantzoulatos we could process the gcode with a script, which modulates the extra restart distance based on the displacement, it would be nice if we could do it inside SuperSlicer! ;-)

tomlop33, 2022-08-30:
Hi, any news or test about this topic ?

thanhtacles, 2023-06-14:
> The Pressure Advance does not solve this problem, it serves another purpose, since it does not dynamically act on the amount of movement of the nozzle. As you can see from my 3DBenchy prints the PA it is well configured, the seams are practically invisible. If every time a retraction and a movement is applied there was something that at the hardware level close the nozzle hole this problem would not exist, but we know well that the hole of the nozzle remains open and at 230° with PLA it is normal for it to lose material. even if it is not extruded. If I perform this print at 200° with the PA activated the result is perfect because the material tends to come out less in the long movements of the nozzle. On the other hand this type of printing with a single perimeter and with a very low infill instead creates other situations that cannot be solved by the Pressure Advance since precisely that too is not dynamic. It is a condition that can be solved by acting on the firmware or slicer, not on the hardware here. Since the firmware has no reason to have this feature it would be useful for it to be in the slicer. A plugin that would analyze the g-code and apply this would also be enough, but I don't think it exists.
> 
> [image]
> 
> [image]

I found this thread because I had the same question. Going to be developing my own dynamic compensation based on part too to see if that works. I think this is the kind of setting that needs to be individualized to specific parts and not the whole print. The reason for some of these gaps is cumulative bottom layer gaps. If there's not enough material, the gaps get bigger even when compensated for with extra length. 
Also what silver glitter filament is this? It looks great @robustini

robustini, 2023-06-16:
@thanhtacles I think this issue maybe needs to be opened on Prusa Slicer, I see the SuperSlicer code stopped for a long time, I don't know if the development will continue, I hope so.
That filament is Form Futura PLA.