supermerill/SuperSlicer #1033: [Feature Request] Combining objects

LaserBee, 2021-03-25:
So I've got two objects I've overlapped on the build plate. It treats them as two separate objects, and "bumps" one over the other. I know I should just take it into Meshmixer or Blender and combine them, but it might be useful if there was an option where two objects overlapping on the build plate would be treated as one. So if I had two cubes arranged so they are half overlapping eachother, the perimeter would follow the outside perimeter, not draw one and then draw the other on top of it. Does this make sense?

neophyl, 2021-03-25:
You can already do that. Th original method is to load your first object, then right click it and select add part. Then select your second model. It will load in your second as part of the first.  They will now be one logical entity.  You can add more as needed. 

In Prusa Slicer 2.3 you can select multiple already loaded objects and right click and select merge. I haven’t actually tried this in Super Slicer yet (and I’m not at my pc at the moment ) but the newer 2.55 builds are built on PS2.3 so it should be accessible on those too.

cropduster, 2021-03-26:
But how are they combined? In ideamaker I can fully rotate and move the parts to connect them at any point I wish and then merge those two objects. In SuperSlicer it seems once they connect, you cannot separate them. Ideamaker is great in that regard.

neophyl, 2021-03-26:
I have just checked with 2.3.55 and the Merge function from the right click is there from PS2.3.  That combines them into a single logical Object just like the older method of loading the second one in as a part of the first.  

You can still manipulate individual parts though as you have previously been able to.  To select just a Part of an Object that contains multiple Parts you need to select it from the tree view on the right.  
If you just try and grab it on the plater it will just grab the whole Object.  

Also note that as long as a single Part of the Object obeys any of the placement rules then the other Parts are not constrained.  By that I mean you can move a Part off the build plate and float it in mid air, which you are not allowed to do with an Object as all Objects need to be connected to the print bed to be printable (usually :)  )

The one thing that doesn't work and that I would class as a bug is the Split options are not available for Merged parts.  Which it really should be.  Especially as if you export the merged object and then load it back in the options do become available.

LaserBee, 2021-03-26:
Neophyl, that's perfect, I didn't realise that! Solved my issue exactly how I wanted it to. Thank you!

closed 2021-06-04 by supermerill

jrpmedia, 2023-10-20:
"Split options are not available for Merged parts"

Has this been resolved yet please?