supermerill/SuperSlicer #1035: First Layer Height Can;t be Greater than Nozzle Diameter Error

AndyEveritt, 2021-03-27:
### Version
2.3.55.5

### Operating system type + version
Windows 10 & Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
E3D Toolchanger. Reprapfirmware 3.2

### Behavior

#### Description
Get "First layer height can't be greater than nozzle diameter" error when using multi extruder printer with different nozzle diameters
[image]

#### Steps to Reproduce
* Set the nozzle diameter for extruder 1 to be larger (eg 0.6 mm), and extruder 2 to be small (eg 0.3 mm)
  * [image]
  * [image]
* Set the first layer height to be a percentage
 * The value must be larger than `100 * (nozzle_diameter_2 / nozzle_diameter_1)%`
  * [image]
* Change the extruder to be used for at least one of the parts on the build plate to extruder 2
* Try to slice

#### Expected Results
The first layer height percentage would be calculated from the nozzle diameter of the extruder that is being used for the specific part.

#### Actual Results
The first layer height percentage is calculated from the nozzle diameter of extruder 1 even if it is not the one being used.

#### Project File (.3MF) where problem occurs

[first layer height error.zip](https://github.com/supermerill/SuperSlicer/files/6215758/first.layer.height.error.zip)

supermerill, 2021-03-27:
yes, it's an unfortunate problem.
in the tooltip I wrote "from default extruder" but I should have written, "from the first extruder".

Problem: I have a collection of extruders used in a print.
what is the one I should take? the bigger one

AndyEveritt, 2021-03-31:
Yeh the tooltip does line up with what is happening but I think what is happening should be changed.

The easiest fix to remove the error from occuring would be to calculate the layer height from the smallest nozzle diameter. However I think the better solution, that would make it more consistent with layer width percentage calculation, would be to calculate each objects first layer height independently.

AndyEveritt, 2021-03-31:
I am not familiar with the technicalities of how PS/SS deals with various layer heigths for different objects but it must support it to some extent due to the variable layer height tool

AndyEveritt, 2021-03-31:
I don't think you can set multiple extruders for a single object in PS but may be wrong 
[image]

If you can, then yes, I think take the smallest.

supermerill, 2021-03-31:
You can, by using a dirrefent extruder for perimeter, infil... also there is the support extruder that is used for the whole plater.
You can also add parts into an object with a different extruder.

supermerill, 2021-03-31:
I'll see if I can grab the information everywhere it's needed.

AndyEveritt, 2021-04-01:
That's a very good point. Using all nozzle diameters used in each object seems to be making the most sense in that case