supermerill/SuperSlicer #1041: Klipper seems still expecting T<x> for Toolchange

linux-paul, 2021-03-28:
Regarding to this topic:
https://github.com/KevinOConnor/klipper/pull/2207
klipper seems still expecting T<x> from the slicer for a tool change operation, am I wrong?

linux-paul, 2021-03-30:
Indeed the docs seem not clear.
We've got a large discussion in the german klipper FB group about this Topic. The doc of the pull request suggest that Kevin just remove the Tx support from the FW to give more flexibility by using it trough a macro. Following this point of view, ACTIVATE_EXTRUDER should not be adressed by the slicer directy. What do you think about the comment from Kevin in the pull request doc?

supermerill, 2021-03-30:
I think that pull request is 2 years old, so the information present here can be very outdated.

meteyou, 2021-03-31:
ACTIVE_EXTRUDER is only to switch for the current extruder stepper all toolchanger or mmu filament change gcode is in T0...Tx. for example the MMU config file. you have only 1 extruder, but you can switch with T0...Tx to the filaments:
https://github.com/KevinOConnor/klipper/blob/master/config/sample-mmu2s-diy.cfg#L221

Kevin deleted the activate_gcode and deactive_gcode, because all this settings are possible to write in the gcode_macro Tx.

meteyou, 2021-03-31:
with your note to use "T[next_extruder]" in the tool changer custom gcode, it is no problem I personally would prefer, when it would be the default tool changer gcode in SuSl...

linux-paul, 2021-04-02:
I think all opinions are pointed out, thank you guys for this discussion. The issue can be closed.

closed 2021-06-14 by supermerill

mtw3d, 2021-08-23:
I know this is closed, but fwiw, this certainly seems to go against the intended behavior for Klipper. The `activate_extruder` gcode is not intended to be called externally. It will work for some very basic dual extruder printers, but most of them require additional setup prior to activating. It's certainly possible to work around that, either as you suggest, or by overriding the default `activate_extruder` gcode in klipper, but the recommended way to do it is to define custom `T0` and `T1` macros, as shown in the multi extruder configuration example

https://github.com/KevinOConnor/klipper/blob/master/config/sample-multi-extruder.cfg

> what is the problem? 

The stock way SS handles this fails when a printer is configured as the Klipper docs suggest. 

> your solution?
 
SS should use `T0` and `T1` (etc.) when changing extruders, rather than the `activate_extruder` command.

supermerill, 2021-08-23:
And a newbie that use klipper and doesn't know that or doesn't care to create the macro won't have a working "default" profile.
If you create the intended klipper macro, set the custom toolchange  gcode to `T[next_extruder]`
Since 2.3.56 if anything is written on it, the auto-generated gcode won't print and so it will replace the 'ACTIVATE_EXTRUDER' by your intended behavior.

meteyou, 2021-08-23:
i personally think that no multi material printer works with only `activate_extruder` and if SuperSlicer would generate `T0`, `T1` as default, it is 1 more hint for newbie that they need this. i personally see it more as a handicap than an advantage...