supermerill/SuperSlicer #1042: One Perimiter on Top Surface adds extra lines...

LaserBee, 2021-03-28:
### Version
2.3.55
### Operating system type + version
Win 10

Changing to "One perimiter on top surface" results in some extra lines. See the two attached photos, one with the feature enabled, one without... 
[image]
[image]

I can't attach the STL, or I would!

neophyl, 2021-03-28:
Known problems with that function in the newer versions.  Just search.

FYI- to attach files they need to be zipped into archives (zip file).  The preferred upload is actually your saved project file (.3mf) as that also contains all your sliced settings and modifiers etc if any as well as the model.  The model on their own are of limited use in duplicating issues.

supermerill, 2021-03-30:
Try with 0 in "One Perimeter on Top Surface" -> "minimum width"

closed 2021-06-14 by supermerill