supermerill/SuperSlicer #1044: Print Settings Not Showing User Presets in UI

oscarspaz, 2021-03-28:
### Version
2.3.55.5

### Operating system type + version
Win10 Pro (19042) 64bit

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega S using Klipper

### Behavior
After Superslicer is opened, previous saved User Print Settings is missing. 
I checked the SuperSlicer-alpha folder. The User Print Settings are saved there. See screenshot

[image]

supermerill, 2021-03-30:
Can you zip your prusaslicer-alpha folder and post it here?

oscarspaz, 2021-04-02:
The fold name under "[user path]/AppData/Roaming/" is "SuperSlicer-alpha", not "prusaslicer-alpha". Could that be the source of the problem. Folder zip attached.

[SuperSlicer-alpha.zip](https://github.com/supermerill/SuperSlicer/files/6250495/SuperSlicer-alpha.zip)

supermerill, 2021-04-03:
these preset are only for 0.4 extruder diameter.
if you also want them for your 0.6 mm, remove the `and nozzle_diameter[0]==0.4` of the print dependancy

You can also check `Show incompatible print and filament presets` in preferences, to get this sorted out.

oscarspaz, 2021-04-03:
> these preset are only for 0.4 extruder diameter.
> if you also want them for your 0.6 mm, remove the `and nozzle_diameter[0]==0.4` of the print dependancy

I see. I started the profile using Anycubic Presets and then save them (printer, printing, and Filament) with new names assuming that they new profiles will be created. Are you saying that I should have started with a blank/custom profile instead of loading and tweaking a Preset Profile?

supermerill, 2021-04-04:
no, just check the print dependency field and edit it to be sure the print profile do what you want.
Some profiles have their width fixed, so they aren't compatible with other extruder diameters.

oscarspaz, 2021-04-10:
Ok. Thank you.

closed 2021-04-10 by oscarspaz