supermerill/SuperSlicer #1046: The first tab of the new GUI should be named 3D editor

liftbag, 2021-03-29:
### Version
_SuperSlicer 2.3.55.5 Commit 80bee28

### Operating system type + version
_macOS Big Sur 11.2.2_

### Behavior
Shouldn't the first tab of the new GUI be named 3D editor?

[image]

JohanAR, 2021-03-29:
The first tab is supposed to be called "3D view" and the second "Sliced preview", which they do on Linux (and probably on Windows which I believe merill is using for development) so this could be a Mac specific bug. Though on Linux it's impossible to switch between the new tabs after slicing a model, does this happen on Mac too?

qwewer0, 2021-03-29:
Looks good on my end. (Windows)
[image]

liftbag, 2021-03-31:
> The first tab is supposed to be called "3D view" and the second "Sliced preview", which they do on Linux (and probably on Windows which I believe merill is using for development) so this could be a [...]

No, in the version related to this issue, switching between tabs is possible on the macOS version.
The only flaw, apart from the name of the tabs, is that after slicing the g-code view is displayed in the 3D view tab.
[image]

supermerill, 2021-03-31:
so
the old version
 - works on windows
 - can't switch tab on linux
 - can switch tab on macos but the name is the same on each tab and the gcode preview is on the first tab
the new version
 - works on windows
 - works on linux
 - can't switch tab on macos
that is right?

liftbag, 2021-03-31:
> that is right?

Exactly

JohanAR, 2021-03-31:
I might add that I compiled my own superslicer, in case there were any new commits that came in after nightly was built

liftbag, 2021-03-31:
> can you repeat the process and paste the log here?
> thx.

How can I view that log?

supermerill, 2021-03-31:
If you launch the software from a command line, you should see something in the terminal. At the startup, if should write 
```
create settings  print.ui
create settings  filament.ui
create settings  sla_print.ui
create settings  sla_material.ui
create settings  printer_fff.ui
create settings  extruder.ui
```

liftbag, 2021-04-01:
> If you launch the software from a command line, you should see something in the terminal. At the startup, if should write
>
> ```
> create settings  print.ui
> create settings  filament.ui
> create settings [...]

macOS terminal does not produce any log of the operations that take place in the slicer. Maybe it should be started with some additional parameters?

[image]

liftbag, 2021-04-01:
Ok, this is the log after launching the program with the parameter --loglevel 5.
It refers to Commit b446079 nightly.

[Saved terminal Output.zip](https://github.com/supermerill/SuperSlicer/files/6245388/Saved.terminal.Output.zip)

supermerill, 2021-04-01:
Sorry, i wasn't clear enough (and the build failed anway).
Can you test the commit 85b75a, or the compiled version from https://github.com/supermerill/SuperSlicer/actions/runs/709474189
?

liftbag, 2021-04-01:
> Sorry, i wasn't clear enough (and the build failed anway).
> Can you test the commit 85b75a (it's in the `CI` branch), or the compiled version from https://github.com/supermerill/SuperSlicer/actions/runs/709474189
> ?

The program exits on opening with the following error.

[image]

supermerill, 2021-04-01:
Damn! of course!
can you copy/paste /rename `superslicer/resources/cog.svg` to `superslicer/resources/printercog.svg` and `superslicer/resources/spool_cog.svg` (or pull 446fcc or wait the new build)
and retry?

liftbag, 2021-04-02:
> Damn! of course!
> can you copy/paste /rename `superslicer/resources/cog.svg` to `superslicer/resources/printercog.svg` and `superslicer/resources/spool_cog.svg` (or pull 446fcc or wait the new build)
> and retry?

The new build seems to work fine.
Here is the output from the terminal.

[Saved Terminal Output.zip](https://github.com/supermerill/SuperSlicer/files/6250819/Saved.Terminal.Output.zip)

supermerill, 2021-04-03:
mmm notyhing wrong in this log... i wonder where comes the problem.

Can you try the next build (43f60cc) ? Hoping it will select the right tab this time.

liftbag, 2021-04-04:
> mmm notyhing wrong in this log... i wonder where comes the problem.
> 
> Can you try the next build ([43f60cc](https://github.com/supermerill/SuperSlicer/commit/43f60cc2800914251b17cc5e4ba8f9f0e59192f7)) ? Hoping it will select the right tab this time.
> 
> if i can have some screenshot to better understand what's going on, this would be nice.

Sorry, maybe my message wasn't very clear. The build 43f60cc works as intended, without the problems mentioned in the previous messages. The log comes from that version.
If you need it as a debug, I can re-download the problematic version and produce the log with the sequence of operations you asked for.

liftbag, 2021-04-05:
Meanwhile I opened the latest build from the CI branch (93e5584). There is only the plater tab, like in release, but the icons for switching from 3D view to g-code view have disappeared.

[image]

supermerill, 2021-04-11:
right, bad copy-paste, fixed.

closed 2021-05-27 by supermerill