supermerill/SuperSlicer #1049: Weird gap infill behaviour

Phil1988, 2021-03-30:
### Version
2.3.55.5

### Operating system type + version
Windows 10

### Behavior
I have a part where the infill is a bit weird.
When the part gets smaller, the latest SuperSlicer v2.3.55.5 does make these rectilinear infill patterns, while v2.2.53.3 made a gap infill.
I compared the settings and have no idea how to solve it.

Here is a screenshot:
v2.3.55.5
[image]

v2.2.53.3
[image]

Additionally on a smaller part, it even makes a layer, where no infill at all is placed (hard to see, but the purple recilinear infill is from the layer before):
v2.3.55.5
[image]

on v2.2.53.3 it lookes like this:
[image]

#### Project File (.3MF) where problem occurs for both slicers
[SuperSlicer_comparison.zip](https://github.com/supermerill/SuperSlicer/files/6227001/SuperSlicer_comparison.zip)

supermerill, 2021-03-30:
[image]

maybe already fixed

Phil1988, 2021-03-31:
Well this from you looks good...

Mine still looks like this:
[image]

Im using 2.3.55.5
Any ideas?

supermerill, 2021-04-03:
You can try to download the latest nightly to be sure we have the same version and load the project file you sent me, to be sure we load the same thing.

Phil1988, 2021-04-05:
I tried to start the nighly on 3 different Win 10 x64 PCs without success.

I downloaded the nightly_win64 from here:
https://github.com/supermerill/SuperSlicer/actions/runs/715598743

Anything I am doing wrong?

neophyl, 2021-04-05:
I had the same problem with the latest nightly.  I found that running the "@SLIC3R_APP_CMD@.exe" started it though.  The other normal exe just result in nothing opening.

Phil1988, 2021-04-07:
thanks @neophyl 
I was able to open it that way, but the opened 3mf project did not show anything in the 3D, Sliced or Gcode preview.
[image]

so if I am doing anything wrong or I should check, please let me know @supermerill 

Thanks :)

supermerill, 2021-04-11:
oups. Naming bug.

Phil1988, 2021-05-01:
Hi @supermerill 
Will there be a new release without this bug anytime soon?

I know I can use the nightly, but I would rather use a stable. 
Mainly because the nightly is still a bit buggy like switching automatically to the layer view and sometimes is slower on respond.

supermerill, 2021-05-01:
I have to finish fixing the bugs.

closed 2021-06-26 by supermerill