supermerill/SuperSlicer #1052: Retraction when Tool is disabled -> extra length on restart -> setting is not working

well-engineered, 2021-03-30:
### Version
_2.3.55_

### Operating system type + version
_osX 11.2.3_

### 3D printer brand / version + firmware version (if known)
_E3D Toolchanger - RepRapFirmware for Duet 2 WiFi/Ethernet 2.05.1 (2020-02-09b1)_

### Behavior
* Retraction when Tool is disabled -> extra length on restart -> setting has no effect in generated gcode

No matter what values I put in there, I can't find them in the generated gcode

[image]

E.G. If I put a value like "123456" in the input-field for T0, I can find it 5 times in the generated gcode. But there are 141 pickups of this Tool (searching for T0 results in 141 hits)...

[image]

[image]

#### Project File (.3MF) where problem occurs
[3mf and gcode.zip](https://github.com/supermerill/SuperSlicer/files/6232087/3mf.and.gcode.zip)

laviRZ, 2022-08-13:
@supermerill This is still not working for me. I change the `retract_length_toolchange` setting and it doesn't affect the gcode. Attaching the project file and resulted gcode.
[pikachu_dual_details.zip](https://github.com/supermerill/SuperSlicer/files/9332014/pikachu_dual_details.zip)
.

Ch4rlesB, 2022-08-31:
Still not working. Basically making superslicer UNUSABLE for IDEX as the toolheads just ooze on the side...

closed 2024-02-11 by supermerill