supermerill/SuperSlicer #1053: Improve the printing time when "with sheath around the support" is selected

robustini, 2021-04-03:
### Version
2.3.55.5

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Delta custom

### Behavior
When there are several supports, I mean separate on the bed, and the "with sheath around the support" option is selected, all the sheaths are printed first and then the supports, this involves countless movements of the nozzle around the print.
It would be more logical and faster if both were printed at the same time, not all sheaths first and then supports.

supermerill, 2021-04-03:
Do you have a simple project that show the problem?

robustini, 2021-04-04:
[FUSOLIERA_2.zip](https://github.com/supermerill/SuperSlicer/files/6254057/FUSOLIERA_2.zip)

supermerill, 2022-03-23:
Can you try the nightly? I reworked that code, tell me if it's enough.

robustini, 2022-03-24:
> Can you try the nightly? I reworked that code, tell me if it's enough.

https://github.com/supermerill/SuperSlicer/actions/runs/2025451330

closed 2022-07-14 by supermerill