supermerill/SuperSlicer #1054: G-code preview options not displaying correctly on Mac OS

timj79, 2021-04-03:
On the latest nightly build for Mac OS, the view option aren't displaying correctly. Only the first letter is displayed instead of the description as shown in the screen shot.

[image]

supermerill, 2021-04-04:
It only does that if the window is too small or even if maximize on a 4k screen?

timj79, 2021-04-04:
It does it both when minimised and maximised. This is on the retina display on a M1 MacBook Pro. 
Feature types is still displaying though

[image]

[image]

I've also noticed the 30% bridge test is doing something strange on the latest version

[image]

supermerill, 2022-03-14:
can't reproduce on latest beta version.

closed 2022-07-14 by supermerill