supermerill/SuperSlicer #1055: Support Interface Material Disabling

doucettom, 2021-04-03:
### Version
2.3.55.5

### Operating system type + version
Xubuntu 20.04

### Behavior
I'm trying to get rid of the arrowed part (see screenshot)
[image]

neophyl, 2021-04-04:
You can’t currently. You can make it solid but you can’t stop it.

dartrax, 2021-06-10:
This was fixed in PrusaSlicer 6f3f362.
It's not in the current stable yet, will hopefully be in the next version.
I hope @supermerill will merge this (soon ;-) ), because it's so hard to remove from build plate right now (PETG on PEI sheet for example...) that I can't use supports in Prusa at all.

dartrax, 2021-09-03:
This is fixed in PrusaSlicer 2.4.0-alpha1

- Using the new "Snug" support style preference eliminates unneccessary overhanging of support which resulted in support structure on build platform as seen in the picture of @doucettom
- Using the new "First layer density" and "first layer expansion" preferences allow to reduce impact of those first layer support structures

neophyl, 2022-02-23:
Its coming guys, Merill is working on the 2.4.1 merge in.  You can download an early version already but its a bit raw at the moment.  Still many issues so.....

doucettom, 2022-03-04:
I'll still give it a spin and let you guys know what I find ! :)

closed 2022-07-14 by supermerill