supermerill/SuperSlicer #1056: Assertion fail in master

gkatev, 2021-04-04:
### Version
Version 2.3.55, latest master (#ebfaadd), from AUR's superslicer-git.

### Operating system type + version
Arch Linux

### Behavior
Hi,
As soon as I open an STL, superslicer encounters an assertion fail:
```
ASSERT INFO:
./src/unix/glx11.cpp(584): assert ""xid"" failed in SetCurrent(): window must be shown

BACKTRACE:
[1] wxGLContext::SetCurrent(wxGLCanvas const&) const
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxNotebookBase::SendPageChangedEvent(int, int)
[8] g_closure_invoke
[9] g_signal_emit_valist
[10] g_signal_emit
[11] gtk_notebook_set_current_page
[12] wxNotebook::DoSetSelection(unsigned long, int)
[13] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[14] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[15] wxEvtHandler::TryHereOnly(wxEvent&)
[16] wxEvtHandler::ProcessEventLocally(wxEvent&)
[17] wxEvtHandler::ProcessEvent(wxEvent&)
[18] wxEvtHandler::ProcessPendingEvents()
[19] wxAppConsoleBase::ProcessPendingEvents()
[20] wxApp::DoIdle()
[21] g_main_context_dispatch
[22] g_main_loop_run
[23] gtk_main
[24] wxGUIEventLoop::DoRun()
[25] wxEventLoopBase::Run()
[26] wxAppConsoleBase::MainLoop()
[27] wxEntry(int&, wchar_t**)
[28] __libc_start_main
```
I tried 3-4 different STLs, and it happened will all of them. I've attached one of them for completeness' sake:
[Brace_Back.stl.zip](https://github.com/supermerill/SuperSlicer/files/6253803/Brace_Back.stl.zip)

I'm not sure how helpful the error message is, please let me know if I could/should check something else for more useful information.

supermerill, 2021-04-04:
thanks for the report.
I'm trying to do some gui stuff and I'm probably doing something bad.
I changed the problematic code in the meantime with a better version, so it will be hopefully corrected in a day or two.

codingcatgirl, 2021-04-04:
I appreciate you responding to issues so quickly. It feels like superslicer is a bit in between-versions-limbo in some way right now, but seeing you work on it makes it far less frustrating :)

supermerill, 2021-04-04:
> It feels like superslicer is a bit in between-versions-limbo

Yes, sadely.
linux & macos specific issues are slowing everything a fair bit for me.

gkatev, 2021-06-01:
Not sure if this is supposed to have been fixed or not, but I am still getting assertion fails:

git master:
```
ASSERT INFO:
./src/unix/glx11.cpp(584): assert ""xid"" failed in SetCurrent(): window must be shown

BACKTRACE:
[1] wxGLContext::SetCurrent(wxGLCanvas const&) const
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxEvtHandler::ProcessPendingEvents()
[8] wxAppConsoleBase::ProcessPendingEvents()
[9] wxApp::DoIdle()
[10] g_main_context_dispatch
[11] g_main_loop_run
[12] gtk_main
[13] wxGUIEventLoop::DoRun()
[14] wxEventLoopBase::Run()
[15] wxAppConsoleBase::MainLoop()
[16] wxEntry(int&, wchar_t**)
[17] __libc_start_main
```

2.3.56.1:
```
ASSERT INFO:
./src/unix/glx11.cpp(584): assert ""xid"" failed in SetCurrent(): window must be shown

BACKTRACE:
[1] wxGLContext::SetCurrent(wxGLCanvas const&) const
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxNotebookBase::SendPageChangedEvent(int, int)
[8] g_closure_invoke
[9] g_signal_emit_valist
[10] g_signal_emit
[11] g_closure_invoke
[12] g_signal_emit_valist
[13] g_signal_emit
[14] gtk_main_do_event
[15] g_main_context_dispatch
[16] g_main_loop_run
[17] gtk_main
[18] wxGUIEventLoop::DoRun()
[19] wxEventLoopBase::Run()
[20] wxAppConsoleBase::MainLoop()
[21] wxEntry(int&, wchar_t**)
[22] __libc_start_main
```

supermerill, 2021-06-01:
I don't have it, must be specific to your os.
when it's happening?

gkatev, 2021-06-02:
It used to happen when switching to the preview, but only if an object is on the plater. If not, the assertion wasn't triggered when switching.

I tried the Nightly branch (Nigthly actually!) with your recent relevant commit, and it seems that the problem has been fixed! Thank you.

supermerill, 2021-12-11:
please comment if still here.

closed 2021-12-11 by supermerill