supermerill/SuperSlicer #1057: Segfault when changing to preview

gkatev, 2021-04-04:
### Version
Version 2.3.55 (AUR's superslicer 2.3.55.5-3)

### Operating system type + version
Arch Linux

### Behavior
Hi,
Superslicer is crashing on me, with some STLs (not with others), when switching to preview.
(So to be clear, import STL; all good. As soon as a I click to view the printing preview; crash)
G-Code generation still works correctly.

All I really have to offer as a symptom is the coredump, which does not seem all that useful tbh, but maybe someone else might be able to reproduce or chime in.
```
#0  0x00007f634b382079 in  () at /usr/lib/dri/radeonsi_dri.so
#1  0x00007f634b383a38 in  () at /usr/lib/dri/radeonsi_dri.so
#2  0x00007f634adafd62 in  () at /usr/lib/dri/radeonsi_dri.so
#3  0x00007f634a97ed09 in  () at /usr/lib/dri/radeonsi_dri.so
#4  0x00007f634a97bff8 in  () at /usr/lib/dri/radeonsi_dri.so
#5  0x00007f634a97ca7a in  () at /usr/lib/dri/radeonsi_dri.so
#6  0x00007f634aa2b34d in  () at /usr/lib/dri/radeonsi_dri.so
#7  0x00007f634aa2bbfb in  () at /usr/lib/dri/radeonsi_dri.so
#8  0x0000564f4f9c8d77 in  ()
#9  0x0000564f4f77129d in  ()
#10 0x0000564f4f772db6 in  ()
#11 0x0000564f4f95a865 in  ()
#12 0x00007f6363a94dce in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#13 0x00007f6363a95227 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#14 0x00007f6363a955d1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#15 0x00007f6363a9567c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#16 0x00007f6363a95782 in wxEvtHandler::ProcessEvent(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#17 0x0000564f4fa23745 in  ()
#18 0x0000564f4fa0bfab in  ()
#19 0x0000564f4fa0aae4 in  ()
#20 0x00007f6363a94dce in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#21 0x00007f6363a95227 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#22 0x00007f6363a955d1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#23 0x00007f6363a9567c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#24 0x00007f6363a95782 in wxEvtHandler::ProcessEvent(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#25 0x0000564f4fa23745 in  ()
#26 0x0000564f4fa0fe47 in  ()
#27 0x0000564f4fa10400 in  ()
#28 0x0000564f4fa1171c in  ()
#29 0x0000564f4f66cbe3 in  ()
#30 0x0000564f4f66ccb5 in  ()
#31 0x0000564f4f66cf01 in  ()
#32 0x00007f6363a94dce in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#33 0x00007f6363a95227 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#34 0x00007f6363a955d1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#35 0x00007f6363a9567c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#36 0x00007f6363a95782 in wxEvtHandler::ProcessEvent(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#37 0x00007f6363a96423 in wxEvtHandler::ProcessPendingEvents() () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#38 0x00007f6363930180 in wxAppConsoleBase::ProcessPendingEvents() () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#39 0x00007f636339341a in wxApp::DoIdle() () at /opt/wxgtk-dev/lib/libwx_gtk3u_core-3.1.so.4
#40 0x00007f6363393544 in  () at /opt/wxgtk-dev/lib/libwx_gtk3u_core-3.1.so.4
#41 0x00007f6362394ea0 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#42 0x00007f63623e8a49 in  () at /usr/lib/libglib-2.0.so.0
#43 0x00007f6362394503 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#44 0x00007f63626a858f in gtk_main () at /usr/lib/libgtk-3.so.0
#45 0x00007f63633afd86 in wxGUIEventLoop::DoRun() () at /opt/wxgtk-dev/lib/libwx_gtk3u_core-3.1.so.4
#46 0x00007f636396636e in wxEventLoopBase::Run() () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#47 0x00007f6363930e86 in wxAppConsoleBase::MainLoop() () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#48 0x00007f63639aedf7 in wxEntry(int&, wchar_t**) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#49 0x0000564f4f5fa950 in  ()
#50 0x0000564f4f0034f5 in  ()
#51 0x0000564f4efe075e in  ()
#52 0x00007f636201bb25 in __libc_start_main () at /usr/lib/libc.so.6
#53 0x0000564f4effc99e in  ()
```
It seems to be a wxgtk issue, but also, it does not occur in superslicer 2.2.54.2.
[... 40 lines cut ...]
#35 0x00007f3ac24d5382 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df382)
#36 0x00007f3ac24d5e87 _ZN12wxEvtHandler20ProcessPendingEventsEv (libwx_baseu-3.0.so.0 + 0x1dfe87)
#37 0x00007f3ac2372db0 _ZN16wxAppConsoleBase20ProcessPendingEventsEv (libwx_baseu-3.0.so.0 + 0x7cdb0)
#38 0x00007f3ac1eff58a _ZN5wxApp6DoIdleEv (libwx_gtk3u_core-3.0.so.0 + 0x26558a)
#39 0x00007f3ac1eff6b4 n/a (libwx_gtk3u_core-3.0.so.0 + 0x2656b4)
#40 0x00007f3ac1056ea0 g_main_context_dispatch (libglib-2.0.so.0 + 0x53ea0)
#41 0x00007f3ac10aaa49 n/a (libglib-2.0.so.0 + 0xa7a49)
#42 0x00007f3ac1056503 g_main_loop_run (libglib-2.0.so.0 + 0x53503)
#43 0x00007f3ac136a58f gtk_main (libgtk-3.so.0 + 0x1d658f)
#44 0x00007f3ac1f1ab26 _ZN14wxGUIEventLoop5DoRunEv (libwx_gtk3u_core-3.0.so.0 + 0x280b26)
#45 0x00007f3ac23ad37e _ZN15wxEventLoopBase3RunEv (libwx_baseu-3.0.so.0 + 0xb737e)
#46 0x00007f3ac2373766 _ZN16wxAppConsoleBase8MainLoopEv (libwx_baseu-3.0.so.0 + 0x7d766)
#47 0x00007f3ac23f2199 _Z7wxEntryRiPPw (libwx_baseu-3.0.so.0 + 0xfc199)
#48 0x00005637dd7aae9b n/a (prusa-slicer + 0x78fe9b)
#49 0x00005637dd2379f8 n/a (prusa-slicer + 0x21c9f8)
#50 0x00005637dd2212d5 n/a (prusa-slicer + 0x2062d5)
#51 0x00007f3ac0cddb25 __libc_start_main (libc.so.6 + 0x27b25)
#52 0x00005637dd230b1e n/a (prusa-slicer + 0x215b1e)
```

#### Project File (.3MF) where problem occurs
Crashes: [MainBody_175mm_6mm_Fitting_Captive.zip](https://github.com/supermerill/SuperSlicer/files/6253840/MainBody_175mm_6mm_Fitting_Captive.zip)
Does not crash: [Brace_Back.zip](https://github.com/supermerill/SuperSlicer/files/6253841/Brace_Back.zip)

Hope there's enough to go on off, let me know if and how I could help.

supermerill, 2021-04-04:
not much, sadly most of the interesting stuff is behind the ` in  ()`.
Maybe it would be better if crashed with a dedug version of the app.

gkatev, 2021-04-04:
I tried compiling with debug symbols, but the function names are still not visible for some reason:
```
Stack trace of thread 447867:
#0  0x00007f7aeb382079 n/a (radeonsi_dri.so + 0xbe6079)
#1  0x00007f7aeb383a38 n/a (radeonsi_dri.so + 0xbe7a38)
#2  0x00007f7aeadafd62 n/a (radeonsi_dri.so + 0x613d62)
#3  0x00007f7aea97ed09 n/a (radeonsi_dri.so + 0x1e2d09)
#4  0x00007f7aea97bff8 n/a (radeonsi_dri.so + 0x1dfff8)
#5  0x00007f7aea97ca7a n/a (radeonsi_dri.so + 0x1e0a7a)
#6  0x00007f7aeaa2b34d n/a (radeonsi_dri.so + 0x28f34d)
#7  0x00007f7aeaa2bbfb n/a (radeonsi_dri.so + 0x28fbfb)
#8  0x000055e37f6fb7f7 n/a (superslicer + 0xc1e7f7)
#9  0x000055e37f48dc63 n/a (superslicer + 0x9b0c63)
#10 0x000055e37f48e186 n/a (superslicer + 0x9b1186)
#11 0x000055e37f680eec n/a (superslicer + 0xba3eec)
#12 0x00007f7b037c4d9e _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.1.so.4 + 0x1e4d9e)
#13 0x00007f7b037c51f7 _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e51f7)
#14 0x00007f7b037c55a1 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e55a1)
#15 0x00007f7b037c564c _ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e564c)
#16 0x00007f7b037c5752 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e5752)
#17 0x000055e37f7560f0 n/a (superslicer + 0xc790f0)
#18 0x000055e37f73c5c6 n/a (superslicer + 0xc5f5c6)
#19 0x000055e37f73eae1 n/a (superslicer + 0xc61ae1)
#20 0x00007f7b037c4d9e _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.1.so.4 + 0x1e4d9e)
#21 0x00007f7b037c51f7 _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e51f7)
#22 0x00007f7b037c55a1 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e55a1)
#23 0x00007f7b037c564c _ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e564c)
#24 0x00007f7b037c5752 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e5752)
#25 0x000055e37f7560f0 n/a (superslicer + 0xc790f0)
#26 0x000055e37f73f779 n/a (superslicer + 0xc62779)
#27 0x000055e37f73fd36 n/a (superslicer + 0xc62d36)
#28 0x000055e37f740f6c n/a (superslicer + 0xc63f6c)
#29 0x000055e37f39bf83 n/a (superslicer + 0x8bef83)
#30 0x000055e37f39c055 n/a (superslicer + 0x8bf055)
#31 0x000055e37f39c2b6 n/a (superslicer + 0x8bf2b6)
#32 0x00007f7b037c4d9e _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.1.so.4 + 0x1e4d9e)
#33 0x00007f7b037c51f7 _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e51f7)
#34 0x00007f7b037c55a1 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e55a1)
#35 0x00007f7b037c564c _ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e564c)
#36 0x00007f7b037c5752 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.1.so.4 + 0x1e5752)
#37 0x00007f7b037c63f3 _ZN12wxEvtHandler20ProcessPendingEventsEv (libwx_baseu-3.1.so.4 + 0x1e63f3)
#38 0x00007f7b03660180 _ZN16wxAppConsoleBase20ProcessPendingEventsEv (libwx_baseu-3.1.so.4 + 0x80180)
#39 0x00007f7b030c341a _ZN5wxApp6DoIdleEv (libwx_gtk3u_core-3.1.so.4 + 0x31a41a)
#40 0x00007f7b030c3544 n/a (libwx_gtk3u_core-3.1.so.4 + 0x31a544)
#41 0x00007f7b020c4ea0 g_main_context_dispatch (libglib-2.0.so.0 + 0x53ea0)
#42 0x00007f7b02118a49 n/a (libglib-2.0.so.0 + 0xa7a49)
#43 0x00007f7b020c4503 g_main_loop_run (libglib-2.0.so.0 + 0x53503)
#44 0x00007f7b023d858f gtk_main (libgtk-3.so.0 + 0x1d658f)
#45 0x00007f7b030dfd86 _ZN14wxGUIEventLoop5DoRunEv (libwx_gtk3u_core-3.1.so.4 + 0x336d86)
#46 0x00007f7b0369636e _ZN15wxEventLoopBase3RunEv (libwx_baseu-3.1.so.4 + 0xb636e)
#47 0x00007f7b03660e86 _ZN16wxAppConsoleBase8MainLoopEv (libwx_baseu-3.1.so.4 + 0x80e86)
#48 0x00007f7b036dedc7 _Z7wxEntryRiPPw (libwx_baseu-3.1.so.4 + 0xfedc7)
#49 0x000055e37f333f33 n/a (superslicer + 0x856f33)
#50 0x000055e37ecf0253 n/a (superslicer + 0x213253)
#51 0x000055e37eccc8ee n/a (superslicer + 0x1ef8ee)
#52 0x00007f7b01d4bb25 __libc_start_main (libc.so.6 + 0x27b25)
#53 0x000055e37ece8ede n/a (superslicer + 0x20bede)
```

The way I tried to enable the symbols was changing CMAKE_BUILD_TYPE to Debug in the package's build script. 
Specifically, the build part of the script looks like this:
```
cd "$srcdir/SuperSlicer-$_pkgtag"
mkdir -p build
cd build

cmake .. \
	-DCMAKE_C_FLAGS=-g \
	-DCMAKE_CXX_FLAGS=-g \
	-DCMAKE_BUILD_TYPE=Debug \
	-DCMAKE_INSTALL_PREFIX=/usr \
	-DOpenGL_GL_PREFERENCE=GLVND \
	-DSLIC3R_FHS=ON \
	-DSLIC3R_STATIC=OFF \
	-DSLIC3R_WX_STABLE=ON \
	-DSLIC3R_GTK=3 \
	-DSLIC3R_BUILD_TESTS=OFF \
	-DwxWidgets_CONFIG_EXECUTABLE=/opt/wxgtk-dev/bin/wx-config

make
```

Am I doing it correctly?
I will do some further investigation, but in the mean time, any tip is welcome!

codingcatgirl, 2021-04-04:
I'm affected by this on Arch as well, had to downgrade to the previous version for now.

supermerill, 2021-04-04:
I'm sorry, but I never had to build on debug on linux, so I don't know that much.

@lordofhyphens Do you know how to debug in Linux?

evilscientress, 2021-04-04:
@gkatev did you execute the binary that is installed by the package or did you execute the binary in the source tree? If it's the not the later the missing debug symbols are explained by the fact that makepkg strips all ELFs and SO of them during the packaging process.

codingcatgirl, 2021-04-04:
There is a documentation on how to not strip debug symbols here: https://wiki.archlinux.org/index.php/Debug_-_Getting_Traces

I'll see if i can try to get a proper stacktrace.

codingcatgirl, 2021-04-04:
Does this help?

```
Core was generated by `/usr/share/SuperSlicer/superslicer'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007ff3f3b9dfac in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length (this=<optimized out>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:908
908	/build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h: Datei oder Verzeichnis nicht gefunden.
[Current thread is 1 (Thread 0x7ff3b3fff640 (LWP 29280))]

(gdb) bt
#0  0x00007ff3f3b9dfac in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const (this=<optimized out>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:908
#1  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=0x7ff3ac030160, __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
#2  0x000055f9b253a133 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (__str=<optimized out>, this=<optimized out>) at /usr/include/c++/10.2.0/bits/basic_string.h:1366
#3  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (__str=<optimized out>, this=<optimized out>) at /usr/include/c++/10.2.0/bits/basic_string.h:667
#4  Slic3r::GCodeProcessor::apply_config(Slic3r::PrintConfig const&) (this=this@entry=0x7ff3b3ffab08, config=
    ...) at /usr/src/debug/SuperSlicer-2.3.55.5/src/libslic3r/GCode/GCodeProcessor.cpp:543
#5  0x000055f9b256d7a5 in Slic3r::DoExport::init_gcode_processor
    (silent_time_estimator_enabled=@0x7ff3b3ffab00: false, processor=..., config=...)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/libslic3r/GCode.cpp:748
#6  Slic3r::GCode::_do_export(Slic3r::Print&, _IO_FILE*, std::function<void (std::vector<Slic3r::ThumbnailData, std::allocator<Slic3r::ThumbnailData> >&, std::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) (this=0x7ff3b3ff5f50, 
    print=..., file=0x7ff3a4001980, thumbnail_cb=...)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/libslic3r/GCode.cpp:1042
--Type <RET> for more, q to quit, c to continue without paging--
#7  0x000055f9b2574ded in Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodeProcessor::Result*, std::function<void (std::vector<Slic3r::ThumbnailData, std::allocator<Slic3r::ThumbnailData> >&, std::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>)
    (this=this@entry=0x7ff3b3ff5f50, print=print@entry=0x55f9b5e09ca0, path=0x7ff3ac018820 "/tmp/.29210.gcode", result=result@entry=0x55f9b5e0cf38, thumbnail_cb=...)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/libslic3r/GCode.cpp:690
#8  0x000055f9b228b690 in Slic3r::Print::export_gcode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Slic3r::GCodeProcessor::Result*, std::function<void (std::vector<Slic3r::ThumbnailData, std::allocator<Slic3r::ThumbnailData> >&, std::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>)
    (this=this@entry=0x55f9b5e09ca0, path_template="/tmp/.29210.gcode", result=0x55f9b5e0cf38, thumbnail_cb=...) at /usr/include/c++/10.2.0/bits/basic_string.h:186
#9  0x000055f9b29e4cb3 in Slic3r::BackgroundSlicingProcess::process_fff() (this=0x55f9b5e0db30)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/slic3r/GUI/BackgroundSlicingProcess.cpp:149
#10 0x000055f9b29e5e99 in Slic3r::BackgroundSlicingProcess::thread_proc() (this=0x55f9b5e0db30)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/slic3r/GUI/BackgroundSlicingProcess.cpp:284
#11 0x000055f9b29e60a7 in Slic3r::BackgroundSlicingProcess::thread_proc_safe() (this=<optimized out>)
    at /usr/src/debug/SuperSlicer-2.3.55.5/src/slic3r/GUI/BackgroundSlicingProcess.cpp:320
#12 0x00007ff3f37e4ed7 in  () at /usr/lib/libboost_thread.so.1.75.0
#13 0x00007ff3f1253299 in start_thread () at /usr/lib/libpthread.so.0
#14 0x00007ff3f117c053 in clone () at /usr/lib/libc.so.6

```

gkatev, 2021-04-04:
> There is a documentation on how to not strip debug symbols here: https://wiki.archlinux.org/index.php/Debug_-_Getting_Traces

Ah, good to know, thanks.

I was able to get a backtrace with symbols. It does seems quite different from @codingcatgirl's one though.
```
Core was generated by `/usr/share/SuperSlicer/superslicer /home/gkatev/Desktop/MainBody_175mm_6mm_Fitt'.
Program terminated with signal SIGSEGV, Segmentation fault.

warning: Section `.reg-xstate/466589' in core file too small.
#0  0x00007fa974bff079 in ?? () from /usr/lib/dri/radeonsi_dri.so
[Current thread is 1 (Thread 0x7fa97ac86080 (LWP 466589))]
(gdb) bt
#0  0x00007fa974bff079 in  () at /usr/lib/dri/radeonsi_dri.so
#1  0x00007fa974c00a38 in  () at /usr/lib/dri/radeonsi_dri.so
#2  0x00007fa97462cd62 in  () at /usr/lib/dri/radeonsi_dri.so
#3  0x00007fa9741fbd09 in  () at /usr/lib/dri/radeonsi_dri.so
#4  0x00007fa9741f8ff8 in  () at /usr/lib/dri/radeonsi_dri.so
#5  0x00007fa9741f9a7a in  () at /usr/lib/dri/radeonsi_dri.so
#6  0x00007fa9742a834d in  () at /usr/lib/dri/radeonsi_dri.so
#7  0x00007fa9742a8bfb in  () at /usr/lib/dri/radeonsi_dri.so
#8  0x000055d7c2a4641b in Slic3r::GUI::GLModel::render() const (this=0x55d7c6cc5fb0) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GLModel.cpp:133
#9  0x000055d7c26b9734 in operator()(Slic3r::Transform3f const&) const (__closure=0x7ffe67748420, transform=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/3DBed.cpp:134
#10 0x000055d7c26b99e8 in Slic3r::GUI::Bed3D::Axes::render() const (this=0x55d7c6cc5f90) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/3DBed.cpp:151
#11 0x000055d7c26bb5b1 in Slic3r::GUI::Bed3D::render_axes() const (this=0x55d7c6cc5d00) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/3DBed.cpp:353
#12 0x000055d7c26ba99c in Slic3r::GUI::Bed3D::render(Slic3r::GUI::GLCanvas3D&, bool, float, bool, bool) const (this=0x55d7c6cc5d00, canvas=..., bottom=false, scale_factor=1, show_axes=true, show_texture=true)
    at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/3DBed.cpp:256
#13 0x000055d7c297f494 in Slic3r::GUI::GLCanvas3D::_render_bed(bool, bool) const (this=0x55d7c716b690, bottom=false, show_axes=true) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GLCanvas3D.cpp:5070
#14 0x000055d7c296890a in Slic3r::GUI::GLCanvas3D::render() (this=0x55d7c716b690) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GLCanvas3D.cpp:1670
#15 0x000055d7c2aae159 in Slic3r::GUI::Preview::on_moves_slider_scroll_changed(wxCommandEvent&) (this=0x55d7c6f31880, event=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GUI_Preview.cpp:1070
#16 0x00007fa98058bd9e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#17 0x00007fa98058c1f7 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#18 0x00007fa98058c5a1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#19 0x00007fa98058c64c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#20 0x00007fa98058c752 in wxEvtHandler::ProcessEvent(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#21 0x000055d7c26d617c in wxWindowBase::ProcessWindowEvent(wxEvent&) (this=0x55d7c71d0c10, event=...) at /opt/wxgtk-dev/include/wx-3.1/wx/window.h:875
#22 0x000055d7c2acaf0c in Slic3r::DoubleSlider::Control::SetSelectionSpan(int, int) (this=0x55d7c71d0c10, lower_val=0, higher_val=20) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/DoubleSlider.cpp:226
#23 0x000055d7c2aacab7 in Slic3r::GUI::Preview::update_moves_slider() (this=0x55d7c6f31880) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GUI_Preview.cpp:855
#24 0x000055d7c25201bf in Slic3r::GUI::Plater::priv::update_preview_moves_slider() (this=0x55d7c6cc2400) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/Plater.cpp:4316
#25 0x000055d7c252f386 in Slic3r::GUI::Plater::update_preview_moves_slider() (this=0x55d7c63ae8d0) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/Plater.cpp:6354
#26 0x000055d7c2a605ce in Slic3r::GUI::GCodeViewer::set_layers_z_range(std::array<unsigned int, 2ul> const&) (this=0x55d7c716bf00, layers_z_range=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GCodeViewer.cpp:675
#27 0x000055d7c29695cc in Slic3r::GUI::GLCanvas3D::set_toolpaths_z_range(std::array<unsigned int, 2ul> const&) (this=0x55d7c716b690, range=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GLCanvas3D.cpp:1857
#28 0x000055d7c2aadf03 in Slic3r::GUI::Preview::on_layers_slider_scroll_changed(wxCommandEvent&) (this=0x55d7c6f31880, event=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GUI_Preview.cpp:1055
#29 0x00007fa98058bd9e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#30 0x00007fa98058c1f7 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#31 0x00007fa98058c5a1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
#32 0x00007fa98058c64c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/wxgtk-dev/lib/libwx_baseu-3.1.so.4
[... cut ...]
#63 0x000055d7c2489284 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/slic3r/GUI/GUI_Init.cpp:84
#64 0x000055d7c18734b0 in Slic3r::CLI::run(int, char**) (this=0x7ffe6774fdd0, argc=2, argv=0x7ffe67750018) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/PrusaSlicer.cpp:569
#65 0x000055d7c1875220 in main(int, char**) (argc=2, argv=0x7ffe67750018) at /tmp/makepkg/superslicer/src/SuperSlicer-2.3.55.5/src/PrusaSlicer.cpp:770
```

codingcatgirl, 2021-04-04:
That is definitely wild and yes, very different. I absolutely can't rule out i did something wrong, c++ isn't my native language, so to speak :D

lordofhyphens, 2021-04-04:
There's a separate assert() that I've been able to trigger when selecting the GCode View w/o having anything on the plate.

lordofhyphens, 2021-04-06:
On this branch over at slic3r/Slic3r: https://github.com/slic3r/Slic3r/tree/merill-merge 

I've tweaked some of the build items so that if you supply `-DCMAKE_BUILD_TYPE=Debug` it'll define wxDEBUG levels and provide `-g` for libslic3r and libslic3r_gui. 

`-DCMAKE_BUILD_TYPE=RelWithDebInfo` will just apply `-g` and not additionally include all  the debug print.

gkatev, 2021-04-14:
FYI there are a bunch of similar-looking issues on the PrusaSlicer repo; https://github.com/prusa3d/PrusaSlicer/issues/6342, https://github.com/prusa3d/PrusaSlicer/issues/5889 to name some.

Somewhere I saw a suggestion of launching with LIBGL_ALWAYS_SOFTWARE=true, which does work for me. The preview is quite laggish with it, but at least it doesn't crash..

supermerill, 2021-06-01:
can you test the latest version?

gkatev, 2021-06-01:
@supermerill By latest, you mean 2.3.56.1 or git master?

gkatev, 2021-06-02:
I tried the Nightly branch and I'd say the issue seems to have been resolved.

On https://github.com/prusa3d/PrusaSlicer/issues/5889 it's been noted that the behaviour is also dependent on the particular model and its comlexity (in some regard).

I don't know to what degree the underlying cause has been fixed or if my success is circumstantial, but I tried a couple different models (somewhat computationally-intensive to slice ones), and Superslicer didn't crash!

supermerill, 2021-06-02:
> I tried the Nightly branch and I'd say the issue seems to have been resolved.

nice!
it's because you're now on the "old" layout or it's really resolved?

gkatev, 2021-06-02:
I'm not sure, I think I'm on the old layout (sliced preview is on the bottom-left). The Nightly branch is the only one I've tested, since others would give me an assertion error (#1056).

closed 2021-06-13 by supermerill