supermerill/SuperSlicer #1058: Print accuracy/serious slicing errors (multiple likely related)

martinpirringer, 2021-04-04:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
2,2,53
### Operating system type + version
_What OS are you using (and version)_
Win 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Anycubic Chiron  1.3.0 stock 

### Behavior
* _Describe the problem_
* Slicing problem # 1 is that it looses the outside single wall perimeter. The outside perimeter thickness is set to .88mm (.8 nozzle) and is ,8 mm Prusa slicer will print it but will mess up on the 3rd wall the 2mm one similare to SS when you set the inside perimeter to 1.5mm where it drops the fill between the 2 .88 perimeters of the 2mm wall and make the 4mm center post hollow
* 
[image]

In both cases (initial case with 1.2 mm inside perimeter walls where it fills between the perimeters on the 2mm wall)
[image]
it drops the outside wall to get it one needs to change the outside perimeter to something less than the .8 it is designed for for example .79
[image]

 

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* To slice it properly
* _Actual Results_
* See pictures
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
  

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2021-04-04:
> Project File (.3MF) where problem occurs

martinpirringer, 2021-04-04:
[TestBox.zip](https://github.com/supermerill/SuperSlicer/files/6254841/TestBox.zip)
Ok I zipped the file. The thin outside wall can be solved by setting thin walls to on. PS prints them by default and well at least its easily fixable. The problem is different from the one you linked to. The problem is that if the inside perimeter seem to be biger than 150ish % of the nozzle it does not fill the gap between the 2 outside perimeters if the wall thickness is slightly thicker than 2x outside perimter like 2mm for an outside perimeter of .88. With an inside perimeter of 1.2 or smaller it will fill that gap all other settings the same. Now in this particular case one could up the outside perimeter but that is not possible when one prints lets say a gear. As that would mess with other things. So I guess the main problem is to make sure the gaps between 2 outside perimeters are always filled if the 2 outside perimeters do not total up to the wall thickness. It is possible as it does it if one decreases the inside perimeter parameter even though in this case no inside perimeter ever is printed