supermerill/SuperSlicer #1061: [Feature] Improve "avoid crossing perimeters" option

robustini, 2021-04-05:
### Version
2.3.55.5

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Delta custom

### Behavior
I notice that when this option is activated the nozzle moves correctly inside the print, but often passes too close to the external perimeters.
This certainly saves time on movement, but in my opinion it is incorrect because in the final result artifacts are noticed if there is a minimum of oozing.
It would be useful to have an option where you can declare how much must be inside the printed area, in mm or in%.
In the image you can see that it passes over that corner, but often in this print it also passes over the long flat front edge. 
The 3mf is the same posted in the issue #1053.

[image]

robustini, 2021-04-05:
Here it is largely evident. 
The red line is how it should be about to do, I hope the concept is clear.

[image]

Pavulon87, 2021-04-05:
Stay inside the infill could be good too.

robustini, 2021-04-08:
Apparently Prusa Slicer has the same problem, so it came from there. 

https://github.com/prusa3d/PrusaSlicer/issues/5689

closed 2021-11-19 by supermerill