supermerill/SuperSlicer #1063: Cant get SuperSlic3r to install on arch

TunjiK, 2021-04-05:
### Version
2.2

### Operating system type + version
Arch

### 3D printer brand / version + firmware version (if known)

### Behavior
`
CMake Error at src/cmake_install.cmake:55 (file):
  file RPATH_CHANGE given FILE "/usr/local/bin/superslicer" that does not
  exist.
Call Stack (most recent call first):
  cmake_install.cmake:107 (include)
`
The folder that it is referring to does exist, not sure what the issue is.

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2021-04-11:
I don't know.

gkatev, 2021-04-12:
Are you building from source? The AUR package's build script may be able to provide insight into required cmake parameters:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=superslicer