supermerill/SuperSlicer #1066: No dpi-awareness

Fivefold, 2021-04-06:
### Version
2.3.55.5

### Operating system type + version
Windows 10 Pro Version 1909 build 18363.1440

Two monitors, one with 150 % dpi-scaling the other with 100 % (no scaling)

### Behavior

Not sure if intended or a bug but SuperSlicer on Windows seems to not be dpi-aware, while PrusaSlicer 2.3.0 is per-monitor-aware (v2). This is on a fresh SuperSlicer install, so I might be missing some setting that I did not see. As far as I remember this worked out of the box with PrusaSlicer.

The missing dpi-awareness leads to either blurry bitmap scaling on one of the two monitors, or if the high DPI scaling behavior is overwritten to be done by the application in the .exe compatibility settings (as described [here](https://support.microsoft.com/en-us/topic/windows-scaling-issues-for-high-dpi-devices-508483cd-7c59-0d08-12b0-960b99aa347d) under 'Workaround' on the third point) the scaling is fixed to the 150 % scaling, leading to the whole UI being too big on the 100 % monitor 

[image]
_("nicht bekannt" means unknown, "Pro-Monitor" means per monitor)_

supermerill, 2021-04-11:
I don't know what i have not done to not have this flag. Most of the code is from prusa...

Also, I have "Per-monitor (V2)" for my dev version of superslicer...
So maybe it's already fixed.

Fivefold, 2021-04-11:
I found the cause. 
I've been launching superslicer.exe through a tool called "[Search Everything](https://www.voidtools.com/)", which itself isn't dpi-aware. For some reason this in turn applies to superslicer too. If I launch regularly through file explorer or a shortcut it will be per-monitor (v2) dpi-aware.

Will close since it doesn't seem to be a problem with superslicer and in any case it should be a rare use-case.

closed 2021-04-11 by Fivefold